openrocket/openrocket #850: Trapezoidal fin sets: default count is "1", should be "3"

neilweinstock, 2020-12-14:
I feel like this was fixed in the past, but it's still showing up.  When creating a new trapezoidal fin set, the default count is "1".  It should be "3".

wolsen, 2020-12-15:
So it looks like the default for all fin types *except* trapezoidal fin set is 3. The trapezoidal fin set count was set to default to 1 with the change for issue #431. Will need to revisit that fix. The fix to default to 3 is trivial, but need to ensure not to re-introduce the issue in 431.

wolsen, 2020-12-17:
Okay, reviewing the file attached to #431 it appears that the issue was with the way the file was saved. The number of fins (instancecount) was not saved within the file when it was saved. Therefore, when the file was opened it defaults to whatever is set as the default. That default was changed down from 3 to 1 to initially fix the issue, however it was the persistence of the instancecount for the finset that actually resolves the problem.

This means that it is likely that there are a few files here and there where the instancecount is not saved as part of the .ork file and opening those up will get the default value. The design should be reconfigured to have the correct number of fins and then saved in order to persist this.

With this, I think the right change is to simply revert the trapezoid default finset count back to 3.

neilweinstock, 2020-12-17:
That sounds reasonable.

PR #851 (2020-12-19): Restore default fin count to 3 for TrapezoidFinSet
TrapezoidFinSet fin count was reduced to 1 in previous commits in order
to fix a bug. The actual problem was that the instancecount was not
persisted in the ork file. This change restores the default to 3.

Fixes #850
commit 312e90b9 (2020-12-19): Restore default fin count to 3 for TrapezoidFinSet

TrapezoidFinSet fin count was reduced to 1 in previous commits in order
to fix a bug. The actual problem was that the instancecount was not
persisted in the ork file. This change restores the default to 3.

Fixes #850
closed 2020-12-19 by wolsen