openrocket/openrocket #852: Add more stage separation options

neutronium95, 2020-12-17:
Some rockets come down in multiple segments, for example a nosecone that comes down on it's own parachute. This can be approximated by putting the nose and the associated parachute in a separate stage, but there isn't a good way to have them separate at apogee besides adjusting it to separate at a certain time in flight. Adding stage separation options for apogee and at a certain altitude during descent would make simulating complicated recovery systems with multiple sections much easier.

RemingtonHolder, 2022-03-22:
I believe this is implemented correct?

JoePfeiffer, 2022-03-22:
No, these options have not been added yet.  Definitely a good idea for the future, though.

PR #2529 (2024-08-12): Add more stage separation options for deployable payloads and a deployable payload example
This adds three new stage separation options: separate at a given altitude (ascending), separate at apogee, and separate at a given altitude (descending). All three can have a delay following the trigger event.

Separate at apogee and separate at altitude (descending) are to support deployable payloads, which descend on their own recovery system separate from the rest of the rocket. Separate at altitude (ascending) is to support a conventional booster/sustainer; I noticed recently that at least one fligh [... cut]
closed 2024-08-12 by JoePfeiffer