openrocket/openrocket #853: New feature request: Ability to add to, remove from, and otherwise modify parts database

hcraigmiller, 2020-12-18:
Allows users to keep their parts database up to date and to create custom parts.

wolsen, 2020-12-19:
This can be done today, but its not exposed via the primary interface.

If you are running using a prepackaged version, just run:
```bash
$ java -cp /path/to/OpenRocket.jar net.sf.openrocket.utils.ComponentPresetEditor
```

Which will require that you open the database file. Its not the most simple to use interface, but its not the worst either. If you have the code checked out and run from an eclipse ide (or other), you an just run the ComponentPresetEditor in the Swing project.

Also note issue #713 where Neil made reference to Dave Cook's database, which is quite extensive. It might be something you'd consider using as well.

neilweinstock, 2020-12-22:
Users should be able to add custom parts without mucking with a database.  Rather, it should work more like motor files, with a small file (JSON, probably) for each part.  Specify a directory and put files in there.

In any part editor dialog, there would be a "Save as custom part" button (or menu item), that saves the part as it currently exists into a file.

Custom part files created in this way could be shared very easily.

As for Dave Cook's parts database, it was previously discussed that we should incorporate it into the new OR release.  I think it would be worthwhile to get it in sooner rather than later, so we can try it out.  I need to go check if there's an issue for this, if not I will file one.