openrocket/openrocket #854: 3D display different than in 15.03

hcraigmiller, 2020-12-19:
Although the 3D display after the recent merge is much better, it is still different than version 15.03.

[image]

In viewing what was intended, I looked back at the "_after_" screen shot in Issue #847 where the display appeared the same as Version 15.03, but the recent merge is not under Windows 10. On my machine with a 32" monitor (not the screen shots above), the 3D display window is completely blank (no text, no rocket), although you can see the rocket in Photo Studio.

hcraigmiller, 2020-12-20:
With regard to no 3D display on my Dell Inspiron with a Samsung 32" monitor, the following is the display information:

[image]

And, the comparison between the current unstable and 15.03 is this:

[image]

I have checked everything I know how to, no joy. I even uninstalled Java 11.0.8 and installed Java 11.0.9, still no joy.

Somehow Version 15.03 is getting it right, and the current unstable not so much. I hope this is helpful.

wolsen, 2020-12-21:
So I think we're ultimately mixing a bunch of issues into the same report here. Various versions of this appear to have been reported in #720 and others. We should try to avoid mixing issues. As I see it, this particular issue is primarily about the scaling of the extra bits beyond just the rocket (altitude, motor selection, CG/CP carets).

I do not have a Windows computer so I will likely need some help debug and troubleshoot this. I have loaded Windows 10 inside a virtual machine but it does not have opengl support so I get no rendering of the 3d views.

wolsen, 2020-12-22:
@hcraigmiller can you check the latest in unstable and report back on what works, what doesn't? I'm hoping the overlay bits (of text and carets) are resolved as well - but haven't been able to recreate these myself.

hcraigmiller, 2020-12-22:
No change in any of the foregoing... should any of the above be spit off into individual issues?

wolsen, 2020-12-23:
Let's split the two issues. I'll relabel this one for 3D renderings in Hi DPI scaling. Please raise a new issue for the lack of 3D renderings on your Samsung display.

wolsen, 2020-12-24:
Okay I think I have this one figured out - but would like you to give it a test @hcraigmiller. I'll send you a jar file to test.

PR #866 (2021-06-04): Fix scaling on high dpi monitors
GLJPanel handles display scaling better than the GLCanvas does natively.
Change to use the GLJPanel always when rendering in 3D mode. FBO
(offscreen rendering) will be set on the GLJPanel via the
GLCapabilities.

Also, ensure when creating a Graphics2D object from a GLOverlay to set
the current transform to the default transform for the component. The
default transform of the swing components will take into account the
scaling of the screen that it is rendered on. By setting the transform
of the overlay, we can be sure that it gets the scaling information as
we [... cut]
closed 2021-06-04 by wolsen