openrocket/openrocket #856: Adjust all freeform fin points color to be red (selected point color to be double red)

hcraigmiller, 2020-12-20:
See comments of neilweinstock in Issue #848.

[image]

wolsen, 2020-12-21:
So I'm not sure why it needs to be red other than "this is how it was in 15.03". It's a minor change, but the question is does this need to change? The grid lines I could easily understand as it was difficult to discern them.

hcraigmiller, 2020-12-21:
Need... no. I believe that you could classify it as one of those "good, better, best" issues where a minor change has a benefit to a user. This was a carry-over issue mentioned by @neilweinstock in Issue #848.

neilweinstock, 2020-12-21:
Having the points be a different color simply makes them easier to see.
The old color choices were good and effective.

It's not a big deal, though; the grid color was much more important from a
usability standpoint.

wolsen, 2020-12-22:
> Having the points be a different color simply makes them easier to see. The old color choices were good and effective. It's not a big deal, though; the grid color was much more important from a usability standpoint.

This is the right reason I think - usability. Will change

PR #858 (2020-12-22): Make points on fin red for usability
Make the points in the fin point scale figure red (once again) to better
show the location of the points.

Fixes #856
commit d8e809c5 (2020-12-22): Make points on fin red for usability

Make the points in the fin point scale figure red (once again) to better
show the location of the points.

Fixes #856
closed 2020-12-22 by wolsen