openrocket/openrocket #859: Mass component radial displacement not shown in 3D view

neilweinstock, 2020-12-22:
Side view:
[image]

3D view:
[image]

hcraigmiller, 2020-12-22:
This issue is confirmed in Window 10 as to all components on the mass row, parachutes, streamers, and mass components of every mass type.

neilweinstock, 2020-12-22:
Ah, interesting.  My guess is that recovery stuff is rarely positioned off-axis, so never noticed before.  There are good reasons to move mass components off center, though.

hcraigmiller, 2021-06-23:
Radial offset is fixed as to all components on the mass row, except a "shock cord"; _the shock cord component is the only mass component that does not have a radial position tab_. 

The lack of a shock cord radial position tab may need to be a new issue.

neilweinstock, 2021-06-23:
I don’t think lack of radial offset on shock cord is an issue.  No one is asking for it.

hcraigmiller, 2021-06-23:
Radial offset of shock cords is commonly done in complex high power rockets. Even though this is of minor significance, I mention it primarily because it is my general belief that all components of the same class should have similar options. Just my view.

neilweinstock, 2021-06-23:
Fair enough. Doesn’t seem high priority, but a reasonable thing to do.

SiboVG, 2021-06-28:
@neilweinstock  This issue was fixed in #957 and merged into unstable, so the issue can be closed.

neilweinstock, 2021-06-28:
Right you are.  Closing...

closed 2021-06-28 by neilweinstock