openrocket/openrocket #860: Conform/Update "Bug report" transmission to location

hcraigmiller, 2020-12-22:
When a user makes a "Bug report" within OpenRocket, the location to which the report is being transmitted should be confirmed/updated.

jimmiedave, 2020-12-22:
Testing on MacOS Catalina 10.15.7, with the Little Snitch network monitor up, upon submission of the report, I see no network activity (no attempt, no communication, no blocking) and an alert is raised:

_"OpenRocket was unable to send the bug report:
Connect Exception: Host is down (connect failed)

Please send the report manually to openrocket-bugs@lists.sourceforge.net"_

**NOTE:** this behavior differs from 15.03 on the same platform. That older version would raise an alert with"

_"IOException: Server responded with code 404, expecting 202"_

neilbalch, 2021-05-11:
Hrm, I'm potentially interested in patching this. Should this somehow hook into a GitHub API for the issues tracker, if one exists? Or should it just direct people to the form to create a new issue and instruct them to copy/paste the bug report text?

PR #893 (2021-05-21): Update Bug report dialog
Updates the Bug report dialog to send users to either raise issues
via the github issues URL for OpenRocket or to send an email to the
forum.

This removes code to auto-submit bugs from the bug report dialog. The
service does not appear to be working and github issues cannot be
anonymous which complicates auto-submission. Instead, just direct users
to the right place.

Fix #860
commit 741db613 (2021-05-21): Update Bug report dialog

Updates the Bug report dialog to send users to either raise issues
via the github issues URL for OpenRocket or to send an email to the
forum.

This removes code to auto-submit bugs from the bug report dialog. The
service does not appear to be working and github issues cannot be
anonymous which complicates auto-submission. Instead, just direct users
to the right place.

Fix #860
closed 2021-05-21 by JoePfeiffer