openrocket/openrocket #861: Photo Studio: "Orbit" background shows up as black

jimmiedave, 2020-12-22:
In both unstable at this writing, and in 15.03, choosing the "Orbit" background (this file in the jar: /datafiles/sky/space.jpg) results in a black background, behind the rendered rocket.

Testing on MacOS Catalina 10.15.7, on a 2020 16" MacBook Pro with graphics acceleration (AMD, I think).

hcraigmiller, 2020-12-23:
Screen shot Windows 10, current unstable.

[image]

Screen shot Windows 10, Version 15,03.

[image]

What operating system is creating the issue?

jimmiedave, 2020-12-23:
As noted above: MacOS Catalina 10.15.7

[image]
The other textures work fine. BTW, they appear to be various panos, where the Orbit shot does not.

[image]
Only difference here is that I've switched to the "Orbit" image.

I've tried a wide variety of monitor settings, thinking that maybe my 4K display ate too much VRAM or something, or that Apple's various 2X HiDPI resolutions might interfere. Nope. Get the same results at 640x480, same results on internal laptop monitor as external 4K.

jimmiedave, 2020-12-23:
After a little investigation, all the textures split on power-of-2 boundaries, except Orbit, which is 1025x1024. I edited it in an image editor to a 1024 square, and it loads fine. Someone needs to trim the image (my trim was a rough experiment) and place it in the JAR, and this will work again.

wolsen, 2020-12-24:
Odd. I don't see the problem in Ubuntu either. Must be limited to MacOS. If cropping the image to 1024x1024 square is all it takes, that's not much.

neilweinstock, 2020-12-28:
Interesting, I have no problem with this in Mojave... on 15.03.  On the latest unstable, Photo Studio is still completely non-functional for me (permanent beach-ball). :(

jimmiedave, 2020-12-28:
Do you have Intel graphics, or a proper GPU? Could make the difference.

DC

David Cain
sblkSoAR@jimmiedave.com
KM4NHQ
NAR L2 #100727
Eggfinder GPS 917MHz, ID3
2020 SoAR VP

jimmiedave, 2020-12-28:
Do you have Intel graphics, or a proper GPU? That could make the difference in both cases.

DC

neilweinstock, 2020-12-28:
I have a 2013 iMac with NVIDIA GeForce GT 750M 1 GB.

Photo Studio worked for me on the unstable branch until I migrated my hard drive from my internal Fusion drive to an external SSD.  Since then... nothing.  OR is the only program that I have identified as being affected.  No idea how to fix.

jimmiedave, 2020-12-28:
Hmm. AMD/Nvidia difference? Driver problem?

Weird.

DC

PR #892 (2021-02-23): Resize space.jpg for display
Resize space.jpg to be 1024x1024 to resolve rendering issues on MacOS.

Fixes #861
commit f71ae2bb (2021-02-23): Resize space.jpg for display

Resize space.jpg to be 1024x1024 to resolve rendering issues on MacOS.

Fixes #861
closed 2021-02-23 by wolsen