openrocket/openrocket #864: 3D display not functioning in Windows !0 an Dell Inspiron with Samsung 3" monitor

hcraigmiller, 2020-12-23:
_On my machine with a 32" monitor (not the screen shots above), **the 3D display window is completely blank** (_no text, no rocket_)_, although you can see the rocket in Photo Studio. With regard to no 3D display on my Dell Inspiron with a Samsung 32" monitor, the following is the display information:

[image]

And, the comparison between the current unstable and 15.03 is this:

[image]

I have checked everything I know how to, no joy. I even uninstalled Java 11.0.8 and installed Java 11.0.9, still no joy.

Somehow Version 15.03 is getting it right, and the current unstable not so much. I hope this is helpful.

hcraigmiller, 2020-12-23:
3D display not functioning in Windows 10 on Dell Inspiron with Samsung 3" monitor split from this issue. New Issue #864.

wolsen, 2020-12-24:
This most certainly will be an issue with opengl not getting initialized properly for one reason or another.

wolsen, 2020-12-24:
If you could run `java -Dopenrocket.debug=True -jar OpenRocket.jar 1> openrocket.log 2>&1` and post the openrocket.log that may help figure out what's going on here

hcraigmiller, 2020-12-26:
When the log was started,  it opened OR. Then, I opened a Simple model rocket, set the vied to 3D Finished, pretended to rotate the rocket, started Photo Studio, rotated the rocket (which was visible, but the background has box lines), closed Photo Studio, and closed OR. Here is the log... 

[openrocket.log](https://github.com/openrocket/openrocket/files/5743428/openrocket.log)

hcraigmiller, 2022-03-03:
User Rob Goring identified a similar issue using Beta 1. He relates that he will provide @JoePfeiffer the log files.

hcraigmiller, 2022-03-18:
Recommend closure in favor of the current Issue #1251.

closed 2022-03-18 by neilweinstock