openrocket/openrocket #865: Fin tabs on tail cone not functioning properly

hcraigmiller, 2020-12-24:
These are 4 related issues pertaining to fin tabs on tail cones [transitions] that are identified as A through D; these can be split into 4 separate issues if materially unrelated. 

Using this design file: 
[ERROR.Transition Fin Tab.zip](https://github.com/openrocket/openrocket/files/5737616/ERROR.Transition.Fin.Tab.zip)

Open and set side view to 150% before viewing the transition and fin root. 

A.  Select the Lower fin set, Fin tab, and click automatic. Two errors are apparent, the fin tab does not go from centering ring to centering ring, and the tab height extends below the inner tube. See "A" below.

[image]

B. Set the Tab position relative to the bottom of the parent component, increase the tab length by 1". Instead of increasing in length toward the top of the component, the tab increases in length toward the bottom of the component and the Tab position is incremented to match. Again, the Tab position should remain unchanged and the tab should increase in length toward the top of the component. See "B" above.

C. When the tab is created manually, the Tab height appears to be the shortest distance between the transition and the inner tube, not the longest distance between the transition and the inner tube that should _always_ be used. See "C" below. 

[image]

D. SAVE the modified design file. RE-OPEN the design file and view the transition and fin root at 150%. The Fin length and Tab position have changed, numerically and visually. See "D" above.

JoePfeiffer, 2020-12-24:
Ah.... when I looked at this before, the fin was only on a single component.  In this case, the fin's parent is the transition, but the fin extends both forward and aft of the transition.  Hmmm...

hcraigmiller, 2020-12-24:
See also #868

hcraigmiller, 2022-08-28:
Fixed by way of PR #1620.

closed 2022-08-28 by hcraigmiller