openrocket/openrocket #871: Automatic base/outer diameter unexpected results/warnings

hcraigmiller, 2020-12-27:
Using the simple model rocket design, _with the nose cone automatic outer diameter off_, turning off the body tube automatic outer diameter causes a body tube discontinuity warning.

[image]

The same is true with the nose cone,  _with the body tube automatic outer diameter off_, turning off the nose cone automatic outer diameter causes a body tube discontinuity warning. This outcome can be changed by manually re-entering the outer diameter for both the nose cone, then the body tube. 

[Opening a new instance of A simple model rocket] Where turning off the automatic body tube outer diameter results in  A below:

[image]

Even though the nose cone shows a base diameter of .984 in. when the design file is opened (see B above), the nose cone may be retaining the default 1.969 in. base diameter when determining body tube discontinuity. This would explain why the body tube discontinuity warning remains after manually entering _the same outer diameter for the body tube as the nose cone base diameter_ (see C above). 

As part of correcting the above, two other suggestions:

1.  **Do not allow** an automatic base/outer diameter box to be checked (grey it out) if there is no other component base/outer diameter to reference. For example, when there is only a nose cone, there is nothing for the automatic base diameter to reference, and the automatic base diameter box should be greyed out. This will prevent A from accidentally becoming B, below:

[image]

2. **Do not allow** _all components_ to be set to automatic base/outer diameter; if this is allowed, there is no "fixed" component base/outer diameter to reference. And, this will prevent A from accidentally becoming B, below:

[image]

hcraigmiller, 2020-12-28:
**Additional information**

If, after you manually re-enter the nose cone base diameter and the body tube outer diameter in the first part above (with automatic sizing for both components off), and the modified design file is saved,  _when the design file is reopened, the same state is preserved_ (the automatic diameter sizing for both components can be toggled on and off without the diameter changing). That file is attached.

[A Simple Model Rocket.Rev.01.zip](https://github.com/openrocket/openrocket/files/5745925/A.Simple.Model.Rocket.Rev.01.zip)

Next, open the design file above, with which you can toggle the automatic diameter sizing of both components on and off without the diameter changing, _set the automatic outer diameter sizing for the body tube to on_, then save the design file. That file is attached.

[A Simple Model Rocket.Rev.02.zip](https://github.com/openrocket/openrocket/files/5745930/A.Simple.Model.Rocket.Rev.02.zip)

When the second design file is reopened, _unchecking the body tube automatic diameter sizing causes the body tube diameter to increase to the default 1.969 in diameter_.

[image]

The previously manually entered body tube outer diameter is lost.

SiboVG, 2021-08-15:
Thank you for the clear bug report @hcraigmiller! Helps fixing the issue by a lot :)

The first three issues that you speak of have to do with the number of decimals in the nose cone diameter and the rounding of that number in the spinner model GUI. In SI units, the nose cone has a diameter of 2.5 cm; in inches this equals 0.984251968... In my case, my units are already in SI units (as should Americans learn to adopt ;) ), so 2.5 cm. If I then in the body tube configuration type in 2.5 cm, I don't get a warning, because this is an exact match with the 2.5 cm of the nose cone. However, when I switched to inches and typed in 0.984, I did get a warning. If you also manually type in 0.984 in the nose cone configuration for its diameter and then try to do the same in the body tube configuration, you don't get an error because you have rounded the 0.984251968... value to 0.984 by manually entering the diameter.

If you think this is confusing and it should be addressed, do you have a proposition of a change that can be made for this? One thought process could be to only let the discontinuity warning come up if the difference is within the 4th or thereafter decimal. However, this is kind of false information and could confuse the user even more if e.g. he/she first enters 0.100 cm for the body tube if the nose cone is 0.101 cm and so a warning comes up and then the user switches to meters for the unit and the warning goes away because you would have 0.001 m for both components.

I agree with your 3 suggestions, will try to implement them so you can test it out.

hcraigmiller, 2021-08-16:
I have experienced the same thing in financial spreadsheets where the sum of a set of numbers is a small amount off because of rounding. The fix with financial maters is to truncate each element to the number of acceptable digits after the decimal point. However, here, that fluctuates with the unit of measure. Looking at SI units, for example, 1112mm is the same as 111.2cm and 1.112m, four digits each. But, OpenRocket currently displays centimeters and meters with one less digit beyond the decimal than what a millimeter entry would be, a discontinuity error just waiting to happen (the actual number of digits beyond the decimal point displayed depends upon the actual length; the larger the number, the fewer digits beyond the decimal displayed.

Again, using SI as an example. the fix is to determine what the smallest acceptable value will be, then scale that up. So, if millimeters were limited to "full" millimeters, then centimeters would only allow 1 digit beyond the decimal and meters only 3; for every digit beyond the decimal for millimeters, you would add one digit to the centimeter and meter digits beyond the decimal shown above. **My recommendation would be 1 digit beyond the decimal for millimeters, 2 for centimeters, and 4 for meters**; no possibility of a rounding error discontinuity.

The more difficult question is how to deal with those pesky "imperial" conversions. The first question is what to convert from and the second is how many digits are "significant" to the conversion. The first seems clear, conversions to and from meters will likely be the cleanest (easiest). As to the second, in school I was always taught that significance was limited to one digit more than the result you wanted. So, the conversion from meters will be 5 significant figures (.33701 for feet and .02808 for inches), rounding the result to 3 digits and 4 digits, respectively.

In other words, _when you change the units of measure, the conversion formula always uses meters as the baseline_, **there is no conversion from inches to feet _or_ feet to inches using 12**. This method should prevent discontinuity errors flowing from conversions between units.

Your thoughts?

SiboVG, 2021-08-22:
PR #998 implements your given suggestions @hcraigmiller. For the rounding of solution: I agree that we can use an absolute bottom reference of e.g. 0.1 mm.

For the bottom reference of non SI units, I would use the following procedure:

- for e.g. the bottom reference of 0.1 mm, to know the bottom reference in inches, I would convert 0.1 mm to inches (which is 0.00393700787 inches), take the most significant digit, and then add one extra decimal to cope with potential rounding errors of the unit conversion. So for inches, results would be rounded up to 0.0001 inches, since 0.003 is the most significant digit and adding one decimal to it, yields 0.0001 inches.

I would work on the other units in the same way: set a bottom limit in SI units and then get the bottom limit of non SI units by the conversion method as mentioned before.