openrocket/openrocket #874: Photo Studio smoke effect does not function

hcraigmiller, 2021-01-02:
Windows 10 OS, Photo Studio, Effects tab. Checking the Smoke box has no visual change, neither does increasing or decreasing the Smoke Opacity, nor changing the smoke color.

neilweinstock, 2021-01-16:
I don't believe it worked in 15.03 either.

neilweinstock, 2021-06-14:
Yup, now working for me on MacOS 11.4 as well.

@hcraigmiller still a problem for you?

hcraigmiller, 2021-06-14:
Using A simple model rocket, with the C6-5 motor... smoke effect is non-functional. All other effects work.

OS Windows 10 with Java version 11.0.9 2020-10-20 LTS.

SiboVG, 2021-06-15:
Tested it on Windows 10 Home, Java Version 8 (1.8.0_291) and it works. Pretty strange...

hcraigmiller, 2021-06-15:
It may be the version of Java. The current unstable is intended for use with Java 11 LTS, and will likely be packaged with that Java version. It is my understanding that the current unstable is not fully compatible with Java 8.

SiboVG, 2021-06-15:
Yeah, but wouldn't the Java 8 version be less compatible then and also have the smoke issue? I don't know, might have a look at it later

SiboVG, 2021-06-15:
I've [read](https://gamedev.stackexchange.com/questions/147218/glsl-shader-error-on-different-computers) about texture2D in swing/resources/datafiles/flame/smokeShader.glsl being deprecated for some GLSL versions. It should be replaced by texture. I've tried replacing this, but a simple replacement isn't enough; other things need to be edited for it to work. If someone wants to work on this, this might be a way to solve it.

The problem might be that hcraigmiller has a different GLSL version (I have no idea, don't know anything about shaders yet, so this is a shot in the dark)

closed 2026-08-24 by SiboVG