openrocket/openrocket #875: RASAero import/export

JoePfeiffer, 2021-01-03:
It would be good to be able to import (and export, though I see that as less important) designs in RASAero's format.

hcraigmiller, 2022-02-21:
After the RASAero II data structure to the current unstable, I believe that it would be appropriate to split this one issue into two issues, a new feature to export an OpenRocket design in the RASAero II file format, and a separate new issue to import RASAero II designs into OpenRocket.

&bull; As to the export new feature, the RASAero II file format is very similar to that of the current unstable, though the variable names are different, and distances are measured in inches, not meters. It appears that all that is needed is a translator that (1) changes the variable names, (2) corrects the numerical values, if needed, and (3) removes excess data fields.

&bull; As to the import new feature, although RASAero II features 8 different fin cross-sections compared to the 3 in OpenRocket, in most other design features RASAero II lacks the component parameters that OpenRocket requires; for example, RASAero does not "position" rail buttons on the body tube, it simply puts 2 buttons on the body tube according to its own internal spacing parameters. This import feature, if even possible, will be much more complex than the export feature.

JoePfeiffer, 2022-02-21:
Breaking it up into two issues seems very reasonable to me, not just because of the difficulties you mention but because the import and export code will be pretty completely separate.

But it's not something I want to think about until after final release of current unstable.

PR #2123 (2023-03-23): [#875] Implement RASAero II importing
This PR fixes #875 and implements RASAero II file (.CDX1) importing. Almost all RASAero components are compatible with OR, although sometimes through the use of inline pods), except for some special stuff like special fin cross sections (see #1626) and protrusions on body tubes. RASAero also uses per-simulation based rocket weight and CG (override).

In short, the following information is imported:
- Rocket components
    - Surface finished are also imported
- Recovery devices
- Simulations
    - Launch site settings are also imported for each sim
    - Mo [... cut]
closed 2023-03-23 by hcraigmiller