openrocket/openrocket #878: Transparency does not display correctly

neilweinstock, 2021-01-16:
@ChrisMickelson (I think) added support for transparency a while back.  I am fairly certain it worked correctly in the past., which is to say that the alpha value was properly reflected in the object appearance in 3D views.  No longer, at least on my Mac running Mojave: now, any alpha value other than 255 displays exactly the same, e.g.:

Color = 0,0,0, 255
[image]

Color = 0,0,0, 254
[image]

Color = 0,0,0, 0
[image]

ChrisMickelson, 2021-01-18:
Thanks Neil, I'll definitely take a look at that. I must have a sixth sense for these things - I have been out of the scene for years and then I get a feeling I should check how OR is doing. I might have to reorient myself to the current state of the project as it seems to have had some serious work done, which is inspiring to see.

neilweinstock, 2021-01-18:
Hey, there you are. :)

I feel certain it used to work correctly, I recall tweaking the alpha channel to get some transparent fins to look just right in a past design.  Definitely misbehaving now, at least on my Mac; haven't been able to confirm on Windows.

wolsen, 2021-02-22:
It took me awhile to confirm on Linux, because I wasn't even aware that there was a alpha/transparency option in the color chooser. Apparently, the Java Swing library's JColorChooser is still missing the transparency selector in the GTK (default) look and feel.

neilweinstock, 2021-02-22:
I was also going to file an issue that we should copy how Rocksim handles transparency: a setting in the "Appearance" tab, along with "color" and "shine".  Having it available only as an alpha channel value inside the color picker is bad.  I realize that it was the easiest way to get it implemented, but from a user standpoint it's far from ideal.

PR #894 (2021-06-22): Fix transparency display issue
Use the provided alpha from the color when it is available when
rendering the model in the 3d finished view. The 3d unfinished view will
continue to override the alpha to show interior components.

Fixes #878
commit ab86cb7f (2021-06-22): Fix transparency display issue

Use the provided alpha from the color when it is available when
rendering the model in the 3d finished view. The 3d unfinished view will
continue to override the alpha to show interior components.

Additionally, enable the GL2.GL_COLOR_MATERIAL for the rendering of
the component to display the right alpha settings.

Fixes #878
closed 2021-06-22 by JoePfeiffer