openrocket/openrocket #881: RockSim coversion error: Distance of fin relative to base/bottom reversed

hcraigmiller, 2021-01-18:
When RockSim measures a distance from the bottom of the component, distances toward the tip (to the left) have positive and distances toward the bottom (to the right) have negative values. In OpenRocket, the reverse is true. OpenRocket correctly inverses the distance for centering rings relative to the base/bottom of the component during conversion...

[image]

But, during conversion, OpenRocket fails to inverse the distance for fins relative to the base/bottom of the component...

[image]

This error results in the fin placement being reversed when relative to the base/bottom of the component.

hcraigmiller, 2021-03-09:
As to this issue, OpenRocket properly converts to RockSim.

jonnypjohnston, 2021-05-24:
I attached a screenshot of the imported file.

I opened the file in Rocksim, saved it out to a different file. Imported that file into OpenRocket.

[image]

PR #920 (2021-05-24): Fix #881, #907 issues
I apologize for the (again) whitespace changes... I can't seem to figure out how to make VSCode not change them. I've tried several user setting overrides, but it keeps changing correcting the whitespace even when I think I'm telling it not to.

I believe this fixes #881. I thought about (_and tried_) changing the `RocksimCommonConstants.ROCKSIM_TO_OPENROCKET_LENGTH` constant with a negative sign, but I soon realized that only the relative position field was bungled, and changing this constant screwed up other parts of the import. I tested it on my own RockSim model, the  [... cut]
commit de10d819 (2021-05-24): Merge pull request #920 from neilbalch/RockSim_Import_Dist_Fix

Fix #881, #907 issues
commit f2b15c10 (2021-05-24): Fix #881 by inverting location value imported from RockSim
closed 2021-05-24 by JoePfeiffer