openrocket/openrocket #885: Show Details"

jimmiedave, 2021-01-31:


hcraigmiller, 2021-02-24:
Post 02/23/2021 merges, this error does not appear using Windows 10. I went back and forth 50+ times between the H100 (DMS) and the H73 (Case) and the window size did not change, nor did I lose the H73 case information. I also tried switching between an assortment of other motors up 20+ times each, both DMS and Case, and SU and Case types, and the window size did not change, nor did I lose the case information on case type motors.

jimmiedave, 2021-02-24:
Same behavior with latest pull from unstable 2/23/2021. Same Mac as above.

Switched to adoptOpenJDK11 in the IDE (and the IntelliJ IDE console/log/whatever indicates this is what's being used this run):

https://youtu.be/jsTyCxU83UQ

jimmiedave, 2021-03-01:
Just confirmed very similar, probably identical, behavior on Ubuntu 20.04.2 LTS, running on AMD Ryzen 3400G, 64-bit Linux.
for test clicks used G69 and G71R.

jimmiedave, 2021-03-07:
In fiddling with 15.03 on Windows 7, I noted that while there are also no resizability controls in either 15.03 or unstable, the panels are never resized/re-layed-out in 15.03. It's possible that the re-sizing in unstable makes the difference old to new. Confirmed that on MacOS 15.03, no resize/layout of panels either.

JoePfeiffer, 2021-03-09:
I'm also seeing similar behavior with latest unstable under Debian Linux, with Java version openjdk 15.0.2 2021-01-19

But... no case information at all in the panel in 15.03?

jimmiedave, 2021-03-09:
git history shows that Kevin Ruland branched in 12/2015, added case info, merged back in 10/10/2016. So that change brings in CaseInfo, and probably the reformatting - apparently that dialog got refactored in the process.

MotorInformationPanel.java is what I'm looking at.

jimmiedave, 2021-03-09:
Think I may have it: simple width of the JPanel is enough to prevent reformatting and remove this problem.

In MotorInformationPanel.java, line 219:

changing:

`			this.add(layer, "width 300:300:, height 180:180:, grow, spanx");`

to

`			this.add(layer, "width 400:400:, height 180:180:, grow, spanx");`

is enough to prevent line-breaks and rewrapping on the longest enums listed in CaseInfo.java - at least on my Mac.

Do I need some kind of permission to checkout, change, ask for test, merge back?

I'm a complete noob at collaborative projects with git, and have only really done Hello World with Java (though I code in a number of languages, know Python, have exposure to C, old 8-bit CPU assembly, lots of others)

JoePfeiffer, 2021-03-09:
If we can find a way to strictly make the size depend on contents, it would be preferable.  It turns out that if we invalidate() the MotorInformationPanel when we update it, it will redraw and fit the contents.  I'm looking at tidying up the compatible cases display as well; PR will be incoming.

git is indeed a strange new world, and I expect to spend a long, long time getting used to git with github's enhancements.

That said, it requires no particular permissions to clone a repository (the closest thing in git to checking it out), create a new branch in your clone, or file a request that your new branch be merged back in to the main line (called filing a pull request).  Actually merging it in is the one thing that requires permissions.

commit 3e3b854c (2021-03-10): Invalidate MotorInformationPanel when updating (fixes #885)

Calls invalidate() as the last action inside MotorInformationPanel.updateData(), forcing a redraw of the panel.  This makes sure the panel is large enough to display all of its contents

Also, clean up display of the compatible cases a little bit by adding a space between cases, and wrapping in <html></html> tags to allow line breaks.
closed 2021-03-10 by wolsen