openrocket/openrocket #886: Clicking on Thrust curve graph in "Show details" pane of Motor selection panel gives scary exception

jimmiedave, 2021-01-31:
Got this upon clicking on the graph (by accident, actually).

Using Unstable
MacOS Catalina 10.15.7, 2020 16in MacBook Pro

[image]

hcraigmiller, 2021-02-23:
Post 02/23/2021 merge, this error does not appear using Windows 10.

jimmiedave, 2021-02-23:
Still happening here on (at least) Aerotech H45 (Total Impulse 289, SU 38x194) and on Aerotech H73.

Noting that the small thrust curve image appears to have two curves for each of these motors, which seems like it might be related to what the exception is describing ("already contains a series...").

hcraigmiller, 2021-02-23:
Specifically looked at the Aerotech H45 (Total Impulse 289, SU 38x194) and the Aerotech H73, with no unusual results using Windows 10. Could this be a Mac issue?

[image]

jimmiedave, 2021-02-23:
Testing on MacOS (same machine as initially listed), the problem occurs with all motors which have both a: no text description (which I note appears in both of your grabs: unusual) and b: 2 plots, up to, but not including J575.

Since you see a text description, there's apparently something in the data that's choking my machine, or my JRE install. Noted also: the pictured are both Tripoli notes. Maybe a commonality? (JRE doesn't like Tripoli‽)

Not sure whether it could be a Mac/Linux/Windows text line-ending thing, or maybe bogon characters embedded in the text. ASCII vs. UTF-8? 

Where's the raw data for these stored? I've had a quick look around but not found much to go on.

JoePfeiffer, 2021-02-23:
The thing with multiple colors is that some (probably most) motors at thrustcurve.org have several curves in the database.  When you select one of these motors, you are given a choice of which curve to use in a dropdown (upper left side of the motor selection dialog).  Each of the possible curves is in a different color (eg red, blue, etc).  When you look at the thrust curve plot, the different curves reflect these options. For some of them, the curves appear identical -- these are probably motors for which there is a rocksim file and a rasp file with identical data (this appears to be the case for a few I've checked), while for others they are radically different (Aerotech H45 comes to mind). When there is information downloaded from thrustcurve identifying the source of a curve, this is described in the text box just above the plot (many have no such documentation).

What *seems* to be going on is that if you've got a motor with more than one thrust curve, and you've selected a curve other than the first, clicking the graph display triggers the exception.  The stack trace in the automated bug report is showing a fault on having more than one curve with a single key; the key is the motor name.

jimmiedave, 2021-02-23:
Ah yes, good detective work. Looks like only the first curve shows up the text in the description too - not sure if that's by accident or that there's a slot for, but no text in, a description per curve > curve 0.

Usability thing: that 'which curve' menu is too far from the curve for most people to associate one with the other. I didn't even know it was there 'til your detailed description. 'fix it someday'. Too much other stuff...

jimmiedave, 2021-02-23:
Just demonstrated the same on 15.03 - Joe, I'm pretty sure you've found it.

PR #901 (2021-03-12): fix typo that called generateSeries() with thrust curve index instead…
… of plot index.

Also added comment explaining necessity of really awkward coding

fixes #886
closed 2021-03-12 by JoePfeiffer