openrocket/openrocket #887: If "Hide motors...not in...production" checked upon starting OpenRocket, they are not filtered.

jimmiedave, 2021-01-31:
Uncheck and recheck filters them.

hcraigmiller, 2021-02-24:
Post 02/23/2021 merge, using Windows 10 with JDK-11, the "Hide motors that are not in regular production" box did not change the Aerotech motors displayed. 

Unchecked the box, saved and reopened, ordered motors by designation, took screenshots of all Aerotech motors displayed. Checked the box and compared displayed Aerotech motors to screenshots, there was no difference. 

Saved with checked box, reopened, ordered motors by designation. Compared displayed Aerotech motors to screenshots, there was no difference. Unchecked the box and compared displayed Aerotech motors to screenshots, there was no difference. 

In all instances, the "Hide very similar thrust curves" and the "Hide motors already used in the mount" boxes were unchecked, as were the boxes limiting motors by dimensions.

jimmiedave, 2021-02-24:
Seeing this behavior in post-02/23/2021 build on MacOS adoptOpenJDK11 JRE as well.

Interestingly, it's apparently not hiding any motors now, including those that are out of production, at any time. Perhaps a change in upstream data or import routines?