openrocket/openrocket #888: Motor data: 29/40-120 casing shows up in modern reloads as 29/40

jimmiedave, 2021-01-31:
Note the case descriptions for G53, G76, G64 and G138. These are all for the 29/40-120 case (I buy and use all of these regularly). The case shows up as 29/40. The outdated reloads below (see prior bug on filtering, whence came this image) show up with the proper designation.

[image]

wolsen, 2021-02-23:
Looks like there must have been an error in the Thrustcurve.org data (I'm guessing maybe?). Just need to update the thrustcurves with the serialize-motors task in the core/build.xml file.

JoePfeiffer, 2021-02-23:
This is probably a a real bug (not just requiring a database refresh).  I've been refreshing it pretty frequently, and I'm seeing the reported behavior. Coincidentally, I was just getting back to the motor stuff...

JoePfeiffer, 2021-02-23:
I apologize, you are correct.

PR #895 (2021-02-23): Update thrustcurves from latest Thrustcurve.org data
Update the thrustcurves from the latest Thrustcurve.org data. This
corrects some of the case information.

Fixes #888
commit 1d0a2c8c (2021-02-23): Update thrustcurves from latest Thrustcurve.org data

Update the thrustcurves from the latest Thrustcurve.org data. This
corrects some of the case information.

Fixes #888
closed 2021-02-23 by JoePfeiffer