openrocket/openrocket #896: Component buttons are too big

neilweinstock, 2021-03-09:
This has been bugging me more and more as time goes on: the component buttons are way too big.  They've gotten considerably larger since 15.03, and as a result far fewer of them fit on the screen at once, and much more scrolling is required to get at them.  Information density is much lower, and it doesn't need to be.

Most of the enlargement is due to much more padding around the perimeter of the buttons.  Possibly, this change was made to accommodate the larger icon images for the new assembly component buttons.  I would argue that those images could be made quite a bit smaller, to fit into the same size buttons as the others, but even if they're left as is, then let those buttons be larger and return the others to their previous size.  No reason to make them all suffer just for those three.

Rocksim doesn't even have icons, which is a bit ugly but has great information density, and all the buttons are visible all the time.  I'm not suggesting we go to that extreme, but right now we're too far in the other direction.

Comparison:
[image]

roguextech, 2021-03-09:
I think about this very same issue <for me any way> every time I use OR. It would be really nice if they were at least two-thirds the size and may be, depending on how much work for y'all, to have an option to not display the text with the icons. The current icon set it quite obvious for the most part any way. Just my 1.5 cents worth..

JP

neilweinstock, 2021-03-09:
I'm always happy to hear that anyone agrees with any of my UI suggestions. :)

An option for icon only could be OK, or even text only.  A three-option selector in Preferences (text, icon, both) covers it.  But even with that, there's still way too much black space in those buttons.

PR #983 (2021-07-16): Insert line break into mass component label
Commit dd277b3 changed the Mass Component label and removed the
line break, which causes the text to no longer wrap and increases
the size of all the buttons. Re-insert the line break to reduce
the component button size.

Fixes #896
commit 120fb72a (2021-07-16): Insert line break into mass component label

Commit dd277b3 changed the Mass Component label and removed the
line break, which causes the text to no longer wrap and increases
the size of all the buttons. Re-insert the line break to reduce
the component button size.

Fixes #896
closed 2021-07-16 by wolsen