openrocket/openrocket #897: Suggestion: keep the "Stage" assembly component button enabled all the time

neilweinstock, 2021-03-09:
Each component button is enabled only when a suitable parent component is selected in the tree.  For all component types that makes sense, except for one: the "Stage" button.  To get it active you need to select "Rocket" in the component tree, which is unusual and (IMHO) not obvious.  Since there's only ever one rocket, there's no ambiguity of where the new stage should go.  So I would just keep that button enabled all the time.

Recall that we removed the always-enabled "New Stage" button from the column of buttons in the middle, which is fine, but the new button should be just as easy to use as the old one.

hcraigmiller, 2021-07-22:
In reviewing the functionality of the component tree, a component button is active only when it would be appropriate to add the component type to be selected to the design. I agree with @neilweinstock that the activation of the _Stage_ component button only when the _Rocket_ is selected is incorrect. However, I disagree that it should always be active, to be consistent with the functionality of the other components, it should be active when either the _Rocket_ or a _Stage_ is selected (the default being named _Sustainer_).

When adding a component, the used may select either a component of the same level, or one level higher. For example, if you have a sustainer stage with a nosecone component, to add a body tube you may select either _Sustainer_ or _Nosecone_ then add the body tube; in both cases the body tube in inserted as the last component in the _Nosecone_ class. And, if there is more than one component of the same class below the component selected, the user is asked whether the component to be added is to be inserted below the selected component or added as the last component of the same class. For example, if you have a sustainer stage with a nosecone and body tube component, and you have the nosecone selected, when adding a second body tube the user is asked whether the second body tube is to be inserted after the nosecone or added to the end. Selecting the sustainer and adding a body tube adds it to the end.

The same structure should be used with stages. If the _Rocket_ is selected when the stage is added, the next stage should be placed at the end.  If the _Sustainer_ (stage) is selected when another stage is added, if there is another stage below the selected stage, then the user should be asked whether to insert or add to the end, otherwise it is added to the end.

The foregoing will bring the addition of stages into conformity with the addition of other components. This should alleviate any confusion for the user as well.

neilweinstock, 2021-07-22:
The problem is this is a regression with regard to the previous version.  There, the stage button was in the middle column, and was always available.  There was never a concept of needing to select "rocket" to add a stage.  Nor, I believe, was there ever a concept of needing to select "rocket" to do anything.

So in essence, we have made the program harder to use in the name of consistency.  In a case such as this I vote for "easy" rather than "consistent".