openrocket/openrocket #898: Enable window to be narrower

neilweinstock, 2021-03-09:
There are many occasions when I want the OR window to be fairly small, so I can have it open side-by-side with either another design(s) or an image editor while i'm working on decals.  Right now, the narrowest the layout can get before it starts clipping is shown here:
[image]

When there are multiple stages its worse, because the stage buttons are laid out horizontally and make the minimum window width even larger.

It looks to me like the limit it being set by the display area.   Within that area, there are two bits of GUI that are consuming excessive horizontal space:  the text "View type" and "Flight configuration".  Arguably "View type" is redundant, since the selected option describes itself as a view.  I'm not sure about "Flight configuration".  Probably need some sort of label for that.

If we can't get rid of the text, then maybe when the window gets sufficiently narrow the text could stack on top of the selector or something?

If I can think of a more coherent suggestion of what I would like to see done, I'll add it here; in the meantime I just wanted to log the issue for future consideration.

hcraigmiller, 2021-03-12:
I agree with @neilweinstock that the information above the display window could be condensed somewhat so that side-by-side programs can be run. This could be done by:

1. Changing the text "View type" to simply "View." Doing so preserves the word "view" for those selections not containing that word (the 3D views) and allows the selection box to be lined up with the view window ruler on the left side.
2. Making the selection/magnification buttons square (shortening the width).
3. Making the distance between the stage button and the percentage selection/magnification button the same as between the percentage display and the percentage selection/magnification button.
4. Changing the text "Flight configuration" to "Configuration"; from a consistency standpoint, the word "flight" should be "motor" (as used on the tab) anyway. To condense the text further, the word "Config" or "Motor" or "Layout" (or another short word) could be used instead of "Configuration." 

These adjustments allow two one-half screens to be side-by-side without clipping.
[image]

neilweinstock, 2021-03-13:
Those are good suggestions.

As a minor side note, "Side view" and "Back view" should really be shortened to simply "Side" and "Back", since the word View is already there on the left.  It doesn't make the window any narrower, because the pull-down width is determined by "3D Unfinished", but it would be a bit more "correct" IMHO.

PR #1381 (2022-05-31): [fixes #898 & #1168] Improve rocket view ribbon for smaller windows + add CG/CP show/hide checkbox
This PR fixes #898 and #1168 by rearranging the rocket view ribbon to be better suited for smaller windows + it adds a checkbox to hide the CG/CP markings in the rocket view. I also added a tooltip text for the rotation slider on the left, since I found it very confusing when I first opened OR what that slider actually did.

Demo:

https://user-images.githubusercontent.com/11031519/170521933-8dddacc5-478f-4f63-b725-0e797a8dabd9.mov

Here is a [jar file](https://github.com/openrocket/openrocket/su [... cut]
commit 2350722d (2022-05-31): Merge pull request #1381 from SiboVG/issue-898

[fixes #898 & #1168] Improve rocket view ribbon for smaller windows + add CG/CP show/hide checkbox
closed 2022-05-31 by JoePfeiffer