openrocket/openrocket #899: Exporting Rocksim file seems broken, and OR gets very confused about the file extension

neilweinstock, 2021-03-09:
Here's what the "Export File" dialog looks like:
[image]
Notice it has given the file the extension ".ork.rkt", rather than simply ".rkt".  This is problem #1.

Then, if I hit "save", here's the file I get:
[image]

It is indeed an ORK file that OR can read in.  So it seems that Rocksim export is totally not working at the moment.

This is on my Mac running Mojave.  It's a pretty recent unstable build, minus the last couple of PRs.

hcraigmiller, 2021-03-09:
Neil, your Issue #899 seems to be the same as Issue #872 which was previously corrected and closed. Can you download the most recent unstable and check to see if the issue still exists?

neilweinstock, 2021-03-09:
Updates to my initial report:

1. I mistakenly said that the exported file was actually an ORK file.  It is indeed a Rocksim file
2. Latest unstable fixes the fact that the exported file has an ORK extension.

**STILL A PROBLEM**: default file name when opening the Export dialog is still <filename>.ork.rkt, so it is still simply adding .rkt to the name rather than replacing the .ork with .rkt.

hcraigmiller, 2021-03-09:
Function in Windows 10 is correct.

[image]