openrocket/openrocket #906: New file opens behind current file

neilweinstock, 2021-03-19:
When opening a file, the window is created *underneath* the current file.  The new file should open on top.

As far as I can tell this is a new behavior, not present in 15.03.

Tested on a Mac running Mojave, Java 11.

hcraigmiller, 2021-03-19:
This does not appear to be a Windows 10 issue...

A simple model rocket was opened first (layering, not full screen), then, from within that window, a second design file was opened. the result:

[image]

In full screen mode, the new design comes up on top, also in full screen mode.

Tested using Windows 10 running Java 11.

JoePfeiffer, 2021-03-19:
Also not seeing it under Debian Linux.  My results match Craig's description.

neilweinstock, 2021-03-19:
Not shocked if this is a Mac-only issue.  Would love to hear confirmation from another Mac user (unfortunately I have only one Mac to try it on).

neilweinstock, 2021-03-25:
This gets weirder.

I have seen this problem on OR 15.03 as well.

Today, it is not happening at all.  I have no theory to explain any of this.  It is one of a number of extremely frustrating bizarre behaviors I've been experiencing lately.

SiboVG, 2021-06-14:
Tested this on my mac (OSX 11.4), and I cannot reproduce this bug.

neilweinstock, 2021-06-14:
I also can no longer reproduce (I'm now also on 11.4).  I'm going to close it for now.

closed 2021-06-14 by neilweinstock