openrocket/openrocket #907: Position component relative to tip of nose cone needs to be brought back

JoePfeiffer, 2021-03-20:
I hadn't noticed it was gone until K'Tesh posted on the mailing list...

neilbalch, 2021-05-13:
Wait, sorry... What does this mean in practice? I'm having a hard time conceptualizing the meaning of this bug report. :(

hcraigmiller, 2021-05-13:
In 15.03, the user has the option of locating an internal part by reference to the "tip of the nosecone." This option does not appear in the current unstable. In designing internal components that are subcomponents of different exterior parts (e.g., two different body tubes), but where the internal components are related to one another in position (e.g., recovery system components that touch or overlap), it is more convenient to position the internal components by referencing the tip of the nosecone instead of the top, bottom, or center of each separate parent component.

neilbalch, 2021-05-13:
Ah. Forgive me if I'm misunderstanding, but shouldn't this be fairly trivial given that the feature already existed? AFAIK we'd just need to `git blame` to find the commit that removed it and then just submit a PR that reverts the changes made... right? :)

hcraigmiller, 2021-05-13:
I'm not familiar with the extend of the coding change(s) that resulted in the loss of the option, but a comparison of the old to the current code sounds like a reasonable first step.

neilbalch, 2021-05-13:
I think my latest PR now addresses this issue.

PR #920 (2021-05-24): Fix #881, #907 issues
I apologize for the (again) whitespace changes... I can't seem to figure out how to make VSCode not change them. I've tried several user setting overrides, but it keeps changing correcting the whitespace even when I think I'm telling it not to.

I believe this fixes #881. I thought about (_and tried_) changing the `RocksimCommonConstants.ROCKSIM_TO_OPENROCKET_LENGTH` constant with a negative sign, but I soon realized that only the relative position field was bungled, and changing this constant screwed up other parts of the import. I tested it on my own RockSim model, the  [... cut]
commit 93633a27 (2021-05-24): Fix #907 by re-adding ABSOLUTE axial offset
closed 2021-05-24 by JoePfeiffer