openrocket/openrocket #908: Frequently can't double-click on a folder in "Open File" and "Save File" dialogs (Mac)

neilweinstock, 2021-03-25:
This cropped up recently, despite having not updated anything of significance on my machine (Mac running Mojave).  It applies to both OR 15.03 and the latest unstable build.

In the file dialogs, if I double-click to navigate into a folder, it just ignores me.  Sometimes, with enough tries, it will eventually work.  Other times I give up and go do something else, because I literally cannot navigate to a file I want to open.  Nowadays it fails in this manner most of the time.

I do not have problems with double-clicking anywhere else, including other other parts of OpenRocket.

This is super weird and extremely frustrating.

wolsen, 2021-03-29:
I wonder if there was a Mac update which caused this? Odd that it applies to both unstable and 15.03

neilweinstock, 2021-03-29:
I really don't know but it is utterly maddening.  OR (old and new) is the only program that has been suffering the problem.

neilweinstock, 2021-03-30:
It seems I am experiencing other double-click problems as well in OR.

Double-clicking a part in the rocket side view is hit or miss at best to bring up the editing panel.  Same with double-clicking a motor to bring up the motor selection dialog.

Geez I hope someone can fix this, or at the very least explain what is going on.

neilweinstock, 2021-04-23:
Follow-up: this is for the mouse only; works fine with Magic Trackpad.