openrocket/openrocket #911: OpenRocket Jar will not launch

roguextech, 2021-04-02:
Ubuntu 20.04
IntelliJ IDEA Community latest version
OpenJava JDK 11 -using
OpenJava JDK 13
Default JRE

Opened project, set JDK to 11, build. Copied jar to desktop. Checked the execute file in the file properties dialog.

Will not run. I've done this a few times before and had no issues.

When I run the project within IDEA, there is a message "failed to load Canberra-gtk-module".

I have reinstalled Ubuntu several times over the past week for other reasons. I always install IntelliJ just for OpenRocket-unstable purposes. This is the first time this has happened.

I can not figure out what is wrong. I have searched high and low using both Google and Bing...

JoePfeiffer, 2021-04-02:
Presumably you have installed the libcanberra-gtk-module package?  On my Debian machine it's 

ii  libcanberra-gtk-module:amd64 0.30-7       amd64        translates GTK+ widgets signals to event sounds

roguextech, 2021-04-02:
Well, I never installed it before and IntelliJ always built without issue.

I would copy the jar to the desktop and she'd run just dandy. I'll install the lib when I get home and report back. About an hourish...

roguextech, 2021-04-02:
Installed the library via:

sudo apt install -y libcanberra-gtk-module-0.30-7

Rebuilt the jar, deleted the old jar, copied the new jar, checked the execute in the file properties dialog, still not running. WTF. I know it is something I did. LOL

I think I am going to delete the src, download a fresh copy ad start over. Just for the heck of it...

JoePfeiffer, 2021-04-02:
Will it run outside IntelliJ using java -jar?  This is the point at which I start doing things like
strace java -jar .../OpenRocket.jar >& log
and looking for strings in the trace...

roguextech, 2021-04-02:
I did that earlier and there was an error. Related to my java installation. Which of course I did not make a copy of. I was having other issues related to Jupyter and reset my Ubuntu notebook.

I will do that again and make notes if it fails...

roguextech, 2021-04-02:
Joe,

See the attached log files. I copied and pasted your CLI example in -01, and just typed it in -02...

[log-01.txt](https://github.com/openrocket/openrocket/files/6251550/log-01.txt)
[log-02.txt](https://github.com/openrocket/openrocket/files/6251552/log-02.txt)

roguextech, 2021-04-03:
I reset my Ubuntu notebook. I am thinking that when I did the ole "sudo apt autoremove" yesterday morning after another reset, that it deleted what ever libraries OpenRocket was looking for.

She is running just fine today. Attached is a screenshot...

[image]

JoePfeiffer, 2021-04-03:
Glad to hear it!   I had actually forgotten a critical option in that shell command, a -f so it would follow forks.  So for the benefit of somebody reading this some time in the future, it should have been
strace -f java -jar .../OpenRocket.jar >& log
But what I'm finding really weird is when I run OR on my Debian box, there's no sign of libcanberra!  So I'm just going to be happy the problem has gone away....

closed 2021-04-03 by JoePfeiffer

roguextech, 2021-04-03:
Noted. Sorry abou that one. However, I do know now not to autoremove extra misc libs...

JoePfeiffer, 2021-04-03:
Not a problem.