openrocket/openrocket #912: Wrong encoding for CSV export

Condzi, 2021-04-12:
When exporting to CSV, units have question marks instead of actual unit. Please, add a possibility to select a proper encoding (so user can select if he/she wants to have questionmarks or actual units, in unicode or utf-8 or something).
The problem: (should be m/s^2)
[image]

neilbalch, 2021-05-11:
Heyo! I'm interested in patching this issue, but I can only seem to reproduce the issue with a plain text editor. _(I tested with VSCode)_ In OR 15.03 on Windows 10 with JRE 1.8.0, Excel reads the CSV in "properly" with the squared character. Could you describe what your computer setup is and what kind of unit representation/format you'd like to see?

[image]

[image]

Condzi, 2021-05-11:
Hello! Thanks for answering.
Unfortunately, I couldn't get it to work with Excel:
[image]
I'm on Windows 10 Pro with following language settings:
[image]
OR 15.03, Java:
[image]
I tried Arial and Consolas fonts, but I still only see the `?`. 
Could you send me the CSV file you're using? Just to see if that's OR issue, or if the issue is on my part. Here's mine (exported from simple rocket example, or something like that)
[csv_files_issue.zip](https://github.com/openrocket/openrocket/files/6456923/csv_files_issue.zip)
> Could you describe what your computer setup is and what kind of unit representation/format you'd like to see?

I would love to see something parser-friendly, maybe just like an additional option when exporting. For example, instead `Lateral acceleration (m/s²)` (which is locale dependent, IIRC), just `ACC_LATERAL METERS_PER_SECOND_SQUARED`. This way the parsing would be easier, I wouldn't have to care about different languages, and I would have just clear structure of `value unit;value unit; value_unit`.

neilbalch, 2021-05-11:
Curious! This is the one I exported from my machine:
[test.zip](https://github.com/openrocket/openrocket/files/6457106/test.zip)

Manually rename it to `test.csv` before opening; GitHub will only let me upload it if it's a `.zip`, lol.

neilbalch, 2021-05-11:
Also, does this one render properly for you?
[test-unicode..zip](https://github.com/openrocket/openrocket/files/6457226/test-unicode.zip)

I manually replaced all of the "squared" characters with the [corresponding unicode superscript](https://en.wikipedia.org/wiki/Unicode_subscripts_and_superscripts) character. I also replaced the degree sign in one of the header items. If this works, I'll try to get a patch PR out soon. _(when my school midterms cool down, lol)_

Also, I noticed that  there are some literal `?` unit designations for a couple of columns at the end: `...Stability margin calibers (?),Mach number (?),Reynolds number (?)...`. Maybe worth investigating whether these are indeed unit-less scalar values or if they do have designations to be added. 🤷‍♂️

neilbalch, 2021-05-11:
> 
> 
> Backing up a step, we've pretty much stopped developing for 15.03 -- all our efforts now are devoted to getting the new release out the door. Are you able to check whether the problem still exists in the current unstable source tree?

Excel reads the header line created by the current unstable branch correctly, just like in 15.03, but VSCode sees the "unicode symbol not found" question mark character. i.e. the issue remains in the current codebase.

neilbalch, 2021-05-11:
> 
> 
> Here's how the `test.csv` from the first comment looks like (different characters than in my file? weird):
> [image]
> But opening it with `Windows 1252` encoding (suggested by VS code` looks correct:
> 
> [image]
> Just to clarify -- opening my own file (directly after exporting to .csv, without opening it in Excel first) still gives `?` instead `²`.
> 
> `test-unicode.csv` looks good in VS Code:
> [image]
> Excel fails to recognize the encoding, I guess:
> [image]
> I think that `?` means a scalar value, since it appears next to coefficients for me. Maybe `scalar` instead `?` would be more fitting there.

Hrm, interesting. I'm actually quite surprised that Excel doesn't read the regular unicode character properly... I'm encouraged by VSCode understanding it, though.

As for the unit-less scalars, I'm in favor of changing them from `?` to `N/A` to indicate that they have no units. This is assuming that the parenthetical unit specification is a mandatory part of the header line template. I haven't found the relevant code yet, so IDK.

PR #919 (2021-05-12): Fix proper UTF-8 encoding for CSV export
This fixes #912 by having `PrintWriter` follow the UTF-8 character set for CSV export.

I found this solution method on Stack Overflow: https://stackoverflow.com/questions/29556253/keep-unicode-characters-in-java-string
closed 2021-05-12 by JoePfeiffer