openrocket/openrocket #914: New feature request: Add file saved indicator to status bar

hcraigmiller, 2021-04-13:
Many programs indicate whether the file has been changed since being opened. For example, the status bar could read:

Unmodified File:  "Rocket Name [File Name (unmodified)]
Modified File:       "Rocket Name [File Name]

I find this feature very helpful in determining whether a file needs to be saved upon exiting, rather than saving every time.

neilweinstock, 2021-04-23:
There's an asterisk next to the rocket name in the title bar if there are unsaved changes.  Are you looking for something more than this?

hcraigmiller, 2021-04-23:
Actually, in all my time using OpenRocket, I never even noticed that asterisk. So, I guess what I am asking is a change to something a little more noticeable and self-explanatory, such as this:

[image]

But, I really do appreciate finally being able to see and understand the asterisk as a saved indicator.

roguextech, 2021-05-10:
Personally for me as well, I like the way it is. That example is nice too though...

loic-fejoz, 2021-05-10:
Using  star character is a pretty common pattern in fact (at least on MS/Windows and some linux app).
See https://ux.stackexchange.com/a/41055 for a more detailed appreciation of the meaning of "common" and other consideration to have.
If someone think to handle this, then much more cases might be handled too (modified on disk, etc) and/or more graphical alternatives might be studied.

SiboVG, 2025-10-08:
> I'll be honest I'm still not convinced any change is needed here.  What did you have in mind?

Same. If we do want to make it clearer, I guess we could add '_(modified)_' after the title when the design is not yet saved. But the current behavior is pretty standard across apps.