openrocket/openrocket #916: When adding new motor configuration, should automatically select the newly-created config

neilweinstock, 2021-04-23:
After creating a new motor configuration, that new configuration should be automatically selected.  The current behavior is weird and a notable regression from 15.03.

Here's the good behavior in 15.03:

First, an existing config is selected:
[image]

Next, a new config is created:
[image]
The new configuration is correctly placed at the bottom of the list, and it is selected.

Here's the behavior in the current unstable:

First, an existing config is selected:
[image]

Next, a new config is created:
[image]
The new configuration is placed at the *beginning* of the list (this time, at least... position is random).  The selection moves up one position to the second-to-last item.  It appears as if the selection is fixed at the sixth row, even when the whole list shifts down.

I suspect this is related to #845, the new random ordering of the motor configurations in the new unstable.

hcraigmiller, 2021-04-23:
This issue may be related to the following issues: #845, #771, and #763.

JoePfeiffer, 2021-04-23:
I suspect you're right.  There's a general problem that these selections aren't synchronized, but should be.  I started to look at it a while ago, but it would be better if someone with more experience with the notifications system took a look.

hcraigmiller, 2021-06-14:
Window 10 Pro...
The Flight configuration window and the motor/recovery configurations are correctly syncing.

But, there is an abnormality when a new configuration is created. Using _A simple model rocket_, create a new motor configuration, then select the new motor (I used an Estes D12-7). You will see the D12-7 in the Flight configuration window, but the flight information (lower left corner) says NA in each of the three information fields. Another motor must first be selected before coming back to the D12-7 for the flight information to be generated.

My hunch is that the command to generate the flight information is somewhere other than at the end of the motor selection routine. But, **this is more of an FYI**... I have found no other anomalies.

PR #942 (2021-06-14): Restore 15.03 selection behavior for flight config
Restore the behavior from the 15.03 version of the flight configuration
panel. Selected flight configuration is synchronized with the rocket
panel where the design is being done and vice versa.

Fixes #916
commit e547c7c3 (2021-06-14): Restore 15.03 selection behavior for flight config

Restore the behavior from the 15.03 version of the flight configuration
panel. Selected flight configuration is synchronized with the rocket
panel where the design is being done and vice versa.

Fixes #916
closed 2021-06-14 by JoePfeiffer