openrocket/openrocket #917: Could we get a new release out soon?

neilbalch, 2021-05-06:
I'm especially interested in using the patch I submitted last year, #556, which brought support for more modern Java JRE/JDK versions.

I moved to a new computer and would like to avoid reinstalling Intellij Idea to rebuild my own version 😛.

neilweinstock, 2021-05-10:
I assure you we wish we could release tomorrow.  Unfortunately wishing won't get us there; there are still a number of issues to resolve.  The good news is we're getting closer; the bad news is it continues to proceed slowly.  It can only come out when it's ready.

The current unstable branch is pretty usable.  I set up a build environment on my Mac so I can update easily from github.  It wasn't too hard and I'm not a Java person at all, I just followed some instructions.

neilbalch, 2021-05-10:
Oh, I'm surprised to hear that... Are the bugs really so numerous and severe that releases need to be paused indefinitely?

I wasn't able to find any instructions for how to set up compiling a `.jar` from the source code. Could you point me to where I could find those?

neilweinstock, 2021-05-10:
They are much less numerous than they were... but it is still not quite ready.

As for building: I worked off the following instructions, although I did it on Mac instead of Windows:
>I first downloaded the openrocket unstable files in a .zip file from GitHub on the Code tab, then placed the files in a directory called "openrocket-unstable." I downloaded "ANT" to its own folder, then added system settings to identify the ANT_HOME as the ANT folder and to place the ANT_HOME \bin folder in the path. Then, from the ANT folder, I first ran ANT, then ran ANT -VERSION to confirm its installation. Then, I changed to the openrocket-unstable folder and ran ANT from within that folder; it constructed the .jar. Apparently, Windows 10 has a limited version of Linux built in, and I had already added Debian, but I don't think either one was necessary to build the .jar file.

And yeah, I needed to fiddle and look up  assorted instructions and whatnot before I got it going, but eventually I got there.

neilbalch, 2021-05-10:
Hrm, interesting. I wasn't able to find any text documentation that mentioned a need to install/configure Apache Ant to compile this project. Noted!

Is there a publically-maintained issue tracker for things that need to be addressed before the next release? AFAIK there are no issue tags that seem to correspond to this.

neilweinstock, 2021-05-10:
There is a summary that is periodically sent to the dev mailing list.

closed 2021-05-11 by neilbalch