openrocket/openrocket #918: Edit Button does not bring up an Edit Window

BadDadio, 2021-05-08:
I am running Openrocket on multiple screens.  It has been suggested to me that my screen configuration has changed since the last time I ran Openrocket.  Yes it has.  I have changed my screen configuration to a single monitor but it the Edit Window still does not appear.  How do I fix this please?

JoePfeiffer, 2021-05-08:
Can you give a clearer picture of just what the sequence of operations is, preferably including the .ork file?  Also, thinking of #906, if you're running on a Mac is it possible the edit window is opening behind the main window?

BadDadio, 2021-05-08:
Running Windows 10.  I open an existing rocket design on one of my three screens.  I click any of the parts in the Rocket design box.  I click the Edit button.  No pop-up window opens anywhere.  I can restore down the Openrocket window and move it around to any screen, but there is no Edit Window behind or on any other screen.  I can change my screens to just 1 or just 2 without ant effect on this.  Thanks for your help.

JoePfeiffer, 2021-05-08:
This is behavior I haven't seen before; if you could upload the .ork file (you'll need to change its extension to .zip to get github to let you upload it) and tell me which component you're trying to edit, I can make a try at duplicating it.

BadDadio, 2021-05-09:
I will upload the file later today.  This suggests that the Edit window memory map is stored is stored in the .ork.  I’m pretty sure that whatever file I open now with my same screen configuration, the same thing happens.

BadDadio, 2021-05-09:
[150K Project BALLS 2020.ork.zip](https://github.com/openrocket/openrocket/files/6448165/150K.Project.BALLS.2020.ork.zip)

BadDadio, 2021-05-09:
I confirmed that none of my .ork files when opened show me the Edit Window.

hcraigmiller, 2021-05-09:
May be related to Issue #864 -- 3D display not functioning in Windows 10 on Dell Inspiron with Samsung 32" monitor.

- Can the Windows 10 edition and version be confirmed [[https://en.wikipedia.org/wiki/Windows_10_version_history](url)].
- Can the Java Version be confirmed.

This may be helpful in identifying the problem.

JoePfeiffer, 2021-05-09:
I was thinking it might be some particular component that somehow isn't hooked up correctly.  If you're seeing the problem with multiple .ork files, Craig is more likely to be able to help you...

neilweinstock, 2021-05-10:
Something I have seen on Windows in the past (but not recently) is that somehow the position of the window gets set to a location that is off the screen.  Since the app remembers the locations of its windows from run to run, once it's offscreen it will continue to open offscreen. until somehow moved back onscreen.  The position of the edit dialog is independent of the main program window, so you can move the program window around all day and the dialog will stay in la-la land forever.

As I recall, this sort of thing usually happened due to display reconfiguration, but was not easy to predict or reproduce.  Sadly I can also not recall exactly how I fixed it.

Where are the window locations and sizes remembered?  Somewhere in the filesystem I presume?  It that info could be cleared out it could fix it.

I cannot guarantee that this is the problem but it sounds a lot like it.

BadDadio, 2021-05-10:
No joy Neil.  It does move the main window to another screen and change the aspect ratio of the window.

Yes, I wish we knew where the window locations were stored.

This did happen to me a few years ago but the Edit window was still barely visible on the right side of the screen where I could click on it and drag it over.

neilweinstock, 2021-05-10:
Rats. :(

neilbalch, 2021-05-11:
> at somehow the position of the window gets set to a location that is off the screen. Since the app remembers the locations of its windows from run to run, once it's offscreen it will continue to [...]

I've always used the method I learned from [this forum post](http://www.virtualdub.org/blog/pivot/entry.php?id=228) to fix the issue. SUPER annoying when it happens, but it's usually pretty simple to recover the window from the ether. _(in my experience)_

BadDadio, 2021-05-11:
Thank you Neil.  I will give it a try in the morning!

Ron

BadDadio, 2021-05-11:
> Oh, cool! So is this issue closed, then?
>
> AFAIK the window-appearing-on-a-now-nonexistent screen is a Windows bug, so not technically OR's fault.

Correct.  Close it.

neilbalch, 2021-05-11:
I think either you or one of the maintainers needs to close the issue. I can't bc I'm not a maintainer. :)

closed 2021-05-11 by JoePfeiffer