openrocket/openrocket #922: New feature request: Provide a detailed wind profile for flight simulations

raketenolli, 2021-05-16:
For high altitude flights, I would like to provide a detailed wind profile (wind speeds and directions as well as their gradients) varying with altitude. These winds would have to be taken into account when determining the flight path and performance in a simulation.

jonnypjohnston, 2021-05-16:
I do not know if this would be beneficial to the developers of OpenRocket, at PSAS we use the [pyHWM2014](https://pypi.org/project/pyhwm2014/) Horizontal Wind Model 2014. I know it is a Python package. Maybe this would be of assistance in developing a wind profile for OpenRocket.

JP

raketenolli, 2021-05-16:
Do I understand correctly that the Horizontal Wind Model provides a wind profile for a given time, latitude and longitude that's based on historic data and measurements? Of course this could be used as a source for wind profiles, but this feature would also include providing a wind profile oneself (e.g. from recent observations) and using it to calculate the 3D flight path.

roguextech, 2021-05-16:
Correct, that is what PSAS is using it for during simulations of our newest launch vehicle. There is quite a bit more info along with SolidWorks models of LV3.1 which is scheduled to launch on July 27, 2021.

The CubeSat that we hope to get sub-orbital within a couple of years is [OreSat](https://github.com/oresat).

Here is the link to the GitHub repo: [LV4-MDO](https://github.com/psas/lv4-mdo)

Charles-McG, 2022-06-27:
I'd like something as simple as the old multi-level wind addon to be functional again.

SiboVG, 2025-02-09:
Fixed by #2559.

closed 2025-02-09 by SiboVG