openrocket/openrocket #925: Bug: Fin tabs move when scaling the rocket

raketenolli, 2021-05-16:
If a fin (I tried trapezoid and freeform) has a fin tab that is offset relative to the "center of the parent component" (which, observing the placement of the tab, is probably not even the right wording – at least in the German translation – because it's relative to the fin root), and then the rocket is scaled to e.g. 200 % its original size, the fin tab moves relative to the fin root by a significant amount:
[image]

I would expect the tab to remain where it was relative to the fin root, with the offset adequately scaled.

hcraigmiller, 2021-05-16:
See also, bugs #865 and #868, and new feature request #829.

hcraigmiller, 2022-01-11:
This issue, Issue #925, resolved with the fix of Issue #868. This issue should be closed.

closed 2022-01-11 by SiboVG