openrocket/openrocket #928: Post "from tip" resurrection: component distance to changed reference point not correctly calculated

hcraigmiller, 2021-05-25:
The following components **DO** correctly calculate the component distance to a changed reference point:
- Trapezoidal fins
- Elliptical fins
- Freeform fins (* excluding conversions from Trapezoidal and Elliptical)
- Tube fins
- Rail buttons
- Couplers
- Centering rings
- Bulkheads
- Engine blocks
- Streamers
- Shock cords
- Mass components

The following components **DO NOT** correctly calculate the component distance to a changed reference point:
- Launch lugs (the distance to a reference point does not change when the reference point is changed)
- Inner tube (incorrect calculated distance to a changed reference point)
- Parachutes (** component reference is the center, not the top or bottom of the component)

_I will post specific screenshots and files to show the errors._

* Conversions from Trapezoidal and Elliptical to Freeform fins _are not_ consistent with the original fin. More testing will be done and further specifics posted.

hcraigmiller, 2021-05-25:
LAUNCH LUGS:
Using the _A simple model rocket_ example file, the distance between the launch lug component and a reference point does not change when the reference point is changed.

[image]

neilbalch, 2021-05-25:
I'm not sure I follow... The render appears to show the correct position in all frames, but the numerical position in the configuration dialog shows `0` for them all. Is the latter the issue at play here in your report?

hcraigmiller, 2021-05-25:
If you were to use a 10" long tube and place a 1" long launch lug in the center of the tube, the reference from the middle would be 0", from the top would be 4.5", and from the bottom would be -4.5". As the screenshots above show, as you move from the middle to any other reference point, the distance does not change. In other words, no matter what reference point you select, the _Middle of the parent component_  is always the actual reference point from which the component distance is calculated and displayed.

hcraigmiller, 2021-05-25:
CONVERSIONS TO FREEFORM FINS
Conversions from Trapezoidal and Elliptical to Freeform fins are not consistent with the original fin dimensions at very small length values, values of > .004".  While this is a bug, it is inconceivable that it would ever be an actual issue in real life. My only concern is that it may evidence a procedural or computational error that goes beyond this manifestation.

neilbalch, 2021-05-25:
I think this is probably due to a separate issue? Not sure, but it sounds distinct from the original position reference bug.

hcraigmiller, 2021-05-25:
If you are referring to the conversion to freeform fins, I agree. I will be separating it out to a new issue for future resolution or simply an FYI in case something related is also found.

SiboVG, 2021-06-15:
See #946, this might have fixed it.

PR #946 (2021-06-26): [fixes #928] Fix launch lug reference
Fixed the issue of the lug distance spinner not updating after another reference point was set. Was fixed by trigger a stateChange on the DoubleModel of the AxialOffset of the lug after an item was selected in the reference JComboBox.

Fixes #928 

Here is the [jar](https://www.dropbox.com/s/zdelttkrufxp3nb/OpenRocket-928.jar?dl=0) for testing.
closed 2021-06-26 by wolsen