openrocket/openrocket #929: Hi-DPI problem in 3D display on Mac

neilweinstock, 2021-05-25:
This may or may not also be a problem on Windows or Linux, hopefully someone else can confirm.  Tested on a pretty recent Unstable build.

Here's the "simple model rocket" in side view on a Retina display:
[image]
Looks good.

Here's the 3D finished view:
[image]
Problems.  The design info in the four corners is behaving as if it doesn't know about hi-DPI display, and the CG/CP markers are off in la-la land.

hcraigmiller, 2021-05-25:
May be related to issues #864, #854, #720, and #719.

wolsen, 2021-06-01:
@neilweinstock any chance you can test with the PR from https://github.com/openrocket/openrocket/pull/866 ?

neilweinstock, 2021-06-03:
Just tried it, looks good!

wolsen, 2021-06-05:
This was was fixed with merge for PR #866

closed 2021-06-05 by wolsen