openrocket/openrocket #930: FYI-Conversions from Trapezoidal and Elliptical to Freeform fins are not consistent with the original fin dimensions at very small length values

hcraigmiller, 2021-05-25:
When the length of Trapezoidal and Elliptical fins is > .004", the conversion to Freeform fins is not consistent with the original fin dimensions; a rectangular fin shape becomes a triangular fin shape.

While this is a bug, it is inconceivable that it would ever be an actual issue in real life. My only concern is that it may evidence a procedural or computational error that goes beyond this manifestation. This is essentially an FYI to keep an eye out for similar type bugs and/or for future resolution, similar to issue #121.

JoePfeiffer, 2021-05-25:
Do you mean < .004"?

hcraigmiller, 2021-05-25:
Correct, sorry for the error.

JoePfeiffer, 2021-06-02:
It would be helpful to document minimum feature sizes OR can handle before anomalies like this turn up.