openrocket/openrocket #939: Numerous problems in 3D view

neilweinstock, 2021-06-11:
These problems are observed on a Mac with M1 processor on Big Sur 11.4.  As far as I can tell, they were all introduced in a recent PR.

1) Load the "Simple model rocket" example, and enable 3D Finished view.   Adjust the position of the rocket with the mouse.  Observed frame rate is now much lower than 15.03 (it is noticeably jumpy).

2) Rotate it so that the nose cone is pointing to the right (180 degrees from default).  On my machine it now looks like this:
[image].  There are two problems here:
a) the original CG and CP markings on the right never moved.
b) new CG and CP markings on the left, in the correct location, are only drawn half.
c) there are some other weird aspects to the CG and CP markings, but hopefully it's all the same problem so I won't try to describe them all.

3) Add a mass component to the nose cone, set it to 4 oz. by using the "up" arrow on the Mass field.  Now looks like this:
[image]
a) The new CG marking has now disappeared.
b) The CG location in the upper right hasn't changed.  It will change to the correct value if you switch to the side view.
c) On my machine, it takes about a second to respond to each "up" click, vs. 15.03 which is nearly instant.

4) After switching back and forth between 3D view and Side view a few times, I was able to get the app to permanently beachball, and I needed to force quit.  I haven't been able to reproduce this yet, but it definitely happened once and should never happen.

hcraigmiller, 2021-06-11:
I ran through all of the problems listed above. I could not replicate any of these problems using Windows 10 Pro, Service Pack 120.2212.2020.0; Java 11.0.9, 2020-10-20, LTS. The software functions correctly in Windows 10 Pro.

wolsen, 2021-06-12:
I've spent some time this evening trying to recreate this issue and I can't recreate it either (Ubuntu 20.10). The sluggishness tells me its something in the graphics stack (i.e. the way jogl is interacting with the 3d renderings or something).  I'll keep playing trying to recreate it.

If you do happen to run into it again, let's try and get a javacore file from it. It may be able to tell us what's going on. Open a command window and find the OpenRocket program (`ps -eaf` or `ps -eaf | grep OpenRocket`). That will give you a process id (pid) which you can then run `kill -9 <pid_num>`. It should generate a javacore file in the location of the OpenRocket jar file.

neilweinstock, 2021-06-12:
I tried this, but I can't find any javacore file...  what would the actual
file name be?

wolsen, 2021-06-14:
I'll get a sample javacore filename in the next day or two.

@neilweinstock just to confirm, you are not able to recreate this at this point in time, correct?

neilweinstock, 2021-06-14:
I was not able to recreate the beachball/hang (#4), but the other problems (#1-#3) are consistent and repeatable.

SiboVG, 2021-08-12:
@neilweinstock  do you have an idea after which PR the issues started showing?

neilweinstock, 2021-08-12:
Only approximately.  We can estimate it's something not too long before the June 11 date when I filed this issue.

Should I tack on the blackout problem here, or file separately?

BTW @SiboVG do you see these problems as well?

neilweinstock, 2021-08-12:
Hmm, that is strange.  Or maybe a driver issue somehow related to the on-board GPU, dunno.  Gonna be hard to debug if you can't reproduce, but good luck. :)

neilweinstock, 2021-08-13:
If these problems are all M1-specific then I wonder if we should remove the beta blocker tag, unless you think you can make any headway on these without having an M1 machine to work on.  Sorry I can't lend you my iMac. :)

SiboVG, 2021-08-13:
Looking at how other systems were unable to replicate the issue(s), I have the feeling that the problems are indeed M1-specific. For that reason we can maybe remove the beta blocker and make it pre-release in the form of Apple M1 Support?

Indeed, debugging without being able to test the problem is kind of a developer's nightmare 🙃 .

My initial thoughts were also that the problems were related to JOGL (Java OpenGL) not supporting M1 well enough, but the fact that some of these issues only started to show up after recent PRs leads me to believe that there is some OR code that we can edit that may alleviate some of the problems.

Just a random thought: do you have Rosetta 2 installed on your mac? If not, could you try installing it and then running the tests again to see if this does any wonders to the issues?

neilweinstock, 2021-08-13:
I do have Rosetta 2, and normally run new the new release under it.  I have an ARM-native JVM installed but it's not integrated into my packaged app (which should probably get updated for some testing before release).

There are a few things I can do that may help here.
1) I have all the old JAR files I've tested with, going back a pretty long way.  I can probably go back and re-test and figure out exactly where the problems started.
2) I can retest these specific issues and see if they also occur with the ARM-native JVM

I will report back when I have some answers.

neilweinstock, 2021-08-14:
I have concluded that these problems are all related to the M1, because I'm now seeing problems 1-3 in all the old builds I have dating back to last October.  I got my M1 iMac toward the end of May, and then a fix for high-DPI displays was merged in early June, and then I noticed the 3D problems.  But they're there in older builds as well.

I don't think these are Rosetta problems, though.  I have a JAR file that @wolsen provided that has native ARM JOGL library; when I run it using the Azul Systems ARM64 JDK, it exhibits the same exact problems.

And so I'm very much inclined to remove the beta blocker tag on this; it should be listed as a known bug for the first beta release.  I'm not exactly sure how anyone is going to find a fix for it until there is a dev with an M1 machine, but we can defer that problem for the moment.

SiboVG, 2021-08-14:
Thanks for the conclusion @neilweinstock! If the next generation of M1 chips comes out, I might get me a new MacBook, but for now, I'd indeed focus on other issues :)

neilweinstock, 2021-11-09:
Come on @SiboVG, you know you want one of those new Macbook Pros. :)

In the meantime, I'm now officially volunteering to do whatever it takes to make progress on this issue.  If no developer has an Apple Silicon Mac then there is no hope for getting this fixed, and it really needs to get fixed at some point.  So I will install whatever tools are necessary and do whatever test and diagnostics are necessary... provided that someone can guide me exactly what to do.  Basically, I can act as someone's remote hands for them.

I don't know if this will actually help, but wanted to put the offer out there.  Waiting until after the beta is fine.

SiboVG, 2021-11-09:
> Come on @SiboVG, you know you want one of those new Macbook Pros. :)

Actually ordered one a month ago, am eagerly waiting for it to arrive this or next week 🤩 (wanted to make it a surprise for you when it arrived, but anyway, now you know that I'll join you as an M1 tester :) )

neilweinstock, 2021-11-09:
Hey, congrats!  I can't wait for you to tell me you don't see the same problems I do. :)

SiboVG, 2021-11-09:
Indeed, finally confirmation whether it's an M1 issue or just you losing your mind

neilweinstock, 2021-11-09:
Why not both?

neilweinstock, 2022-02-28:
K'Tesh seems to be experiencing similar problems with the CP/CG markings on Windows: https://www.rocketryforum.com/threads/announcement-openrocket-version-22-02-public-beta-is-underway-download-it-now.171295/post-2247114.  I don't recall ever seeing this problem on Windows before, will need to do some more testing.

hcraigmiller, 2022-02-28:
Unable to replicate in Windows.
[Windows 11 Pro, Version 21H2, Build 22000.434, Windows Feature Experience Pack 1000.22000.434.0]

PR #1773 (2022-10-28): [#1771, #939, #1191] FIX OVERLAY ISSUES IN 3D VIEW!!!!
This PR fixes #1771, fixes #939, and fixes #1191. No more duplicated CG/CP markings!!!

The issue was that the carets (and extra text, such as stability info - see issue #1771) are drawn on the JOGL Overlay class. The overlay class has a function to mark areas as "dirty", which means they should be repainted. The dirty area apparently wasn't sized properly, which this PR solves.

https://user-images.githubusercontent.com/11031519/198417059-70aaa186-4d41-4315-9d24-3c4438739b72.mp4
closed 2022-10-28 by SiboVG