openrocket/openrocket #947: Can't type in Photo Studio spinners

SiboVG, 2021-06-15:
I can't seem to type any numbers in the spinners of Photo Studio. When typing, nothing happens. I can change the value of the spinner using the up and down buttons, but typing doesn't work. Other spinners, like e.g. 'Shape parameter' of the Nose cone works as it should.

[image]
[image]

hcraigmiller, 2021-06-15:
May be duplicate of Issue #873.

SiboVG, 2021-06-15:
Ah yes, sorry, didn't notice, this issue is irrelevant then

closed 2021-06-15 by SiboVG