openrocket/openrocket #948: Active tabs on Mac look just awful

SiboVG, 2021-06-15:
When a tab is selected and active in a window, on Mac OS this is displayed as a white tab background with white text with just a little bit of a drop shadow. The white text of the active tab should be replaced by e.g. a black color to give it more contrast compared to the background.

[image]

wolsen, 2021-06-16:
Fairly positive this is a Java bug https://bugs.openjdk.java.net/browse/JDK-8251377

wolsen, 2021-07-11:
I've put up a jar file [here](https://www.dropbox.com/s/50joo0qdv88gh0i/OpenRocket-theme.jar?dl=0) which has the options of theme selection using jformdesigner's formdev flatlaf stuff.

@neilweinstock @SiboVG care to try it out to see if it fixes some of the selection stuff. Note, there's a couple of nit issues with it that I'm still looking at, thus no PR of any kind.

SiboVG, 2021-07-11:
> I've put up a jar file [here](https://www.dropbox.com/s/50joo0qdv88gh0i/OpenRocket-theme.jar?dl=0) which has the options of theme selection using jformdesigner's formdev flatlaf stuff.

Well, I really really like the ability to change the theme in the first place!

The Mac OSX theme is no better than system default, but other themes do work and in my opinion look even better than the original. This also adds the possibility for custom themes in the future or dark modes etc.
[image]
[image]

neilweinstock, 2021-07-12:
That is cool, but... I would hope we don't ship it in that form.  There are way too many options, and many of them are either (a) horrible, or (b) too similar to each other.  I don't think that's adding value.

*Especially* since they don't generally solve the problem at hand.  Here's the Motors and Configuraitons tab on either Cyan or Gray (lost track):
[image]

The dark highlight is still being used against black text.  This was true of other themes as well, but I didn't fully examine them all (again, list too long).

wolsen, 2021-07-14:
@SiboVG if you want to take a look at resolving this issue (since you have a Mac and I don't). There's a similar reported issue against the JDK at https://bugs.openjdk.java.net/browse/JDK-8257595, which if you see there's a work around by using UIManager.put("TabbedPane.foreground", Color.black). You can drop that in GUIUtil.setBestLAF for the MacOS case. While looking at it, I suspect you'd be able to fix #968 in a very similar vein.

SiboVG, 2021-07-14:
> @SiboVG if you want to take a look at resolving this issue (since you have a Mac and I don't).
Yup, will give it a try

SiboVG, 2021-07-15:
@wolsen can confirm that this solves the issue.

I do however very much like the idea of custom themes, so I would like to see something like that implemented.

neilweinstock, 2021-07-15:
> @wolsen can confirm that this solves the issue.

Did you check the motor selection/configurations tab?

SiboVG, 2021-07-15:
> > @wolsen can confirm that this solves the issue.
> 
> Did you check the motor selection/configurations tab?

Yes, here is an example:
[image]

neilweinstock, 2021-07-15:
Which theme is that?  I couldn't find any that look like that (which is to say, pretty Mac-looking, with proper tab bar)

In any case, what I have seen is that even when the selected motor looks good (white text on dark background), the main Motors and Configurations tab still looks like this ("Gray" theme):
[image]

wolsen, 2021-07-16:
> I do however very much like the idea of custom themes, so I would like to see something like that implemented.

I do too and can polish that up. But honestly, for now, I'm going to switch my attention to higher priority issues for a beta drop. There's a variety of little nits that need to be handled throughout. I pitched it up as an idea before I realized the Mac bits can be fixed with a relatively small change.

> Which theme is that? I couldn't find any that look like that (which is to say, pretty Mac-looking, with proper tab bar)

That should be the actual Mac theme, just with a fix to the Java UIManager settings. Does not require the theme changes jar I provided.

> 
> In any case, what I have seen is that even when the selected motor looks good (white text on dark background), the main Motors and Configurations tab still looks like this ("Gray" theme):

Yeah, for the sake of this issue - let's discard the theme bits and go with where @SiboVG is looking (making some tweaks without themes to fix the coloring on Mac).

PR #984 (2021-07-17): [fixes #948] Fix bad appearance of active tabs on Mac
This PR fixes #948 where the foreground of active tabbed panes was white on OSX machines, making it very much unreadable. All credits for the solution go to @wolsen .

Here is a [jar file](https://www.dropbox.com/s/rtv7vj30nwfgr67/OpenRocket-948.jar?dl=0) for testing.
commit 49a5d920 (2021-07-17): [fixes #948] Added black foreground for tabbed panes
commit 67929185 (2021-07-17): [fixes #948] Only use change for OSX
closed 2021-07-17 by wolsen