openrocket/openrocket #960: New feature request: add ok/cancel button to component config dialog

SiboVG, 2021-06-25:
Right now when e.g. adding or editing a component, the config dialog only has a 'Close' button. However, I find it more intuitive to have an 'Ok' and 'Cancel' button. The 'Ok' button would kinda serve as the current 'Close' button, i.e. applying the changes that were made to the component. The 'Cancel' button would allow you to discard the changes that you made in the config dialog. This can be accompanied with a confirmation popup dialog of whether the user really wants to discard the changes. This can also be useful when you added a component, but don't really wanted it. Instead of having to delete that components afterwards, you could just hit cancel on the dialog that popped up when you added the component.

JoePfeiffer, 2021-06-25:
Related, while looking at PR 951, I noticed the Photo Studio and its controls don't even have Close buttons!

While you're exactly right on OK/Cancel being a good idea, I'd suggest it can wait until after release.

neilweinstock, 2022-07-11:
> Right now when e.g. adding or editing a component, the config dialog only has a 'Close' button. However, I find it more intuitive to have an 'Ok' and 'Cancel' button. The 'Ok' button would kinda [...]

I discussed this elsewhere, but will repeat here.

Right now, I believe "OK" is not appropriate for us here because it is a _different function_ than one normally associates with OK, which is: "apply changes and close dialog".  In our case, we are applying and committing the changes live as we go, so OK is not required to commit them.  And so, "Close" is a perfectly descriptive name for what the button does, which is just to close the dialog and do nothing else.  The Red X in the upper right does the exact same thing.  Duplication of function here is not a problem; I've seen it in plenty of other places and it's fine.  The red X close widget is there because users expect it to be there, and the CLOSE button is there because users expect some sort of way to dismiss a dialog in the bottom right.  So we have both.

A big benefit of the current approach of committing changes immediately is that you can go from component to component quickly, without extra steps to commit.  In some ways, the current config dialog behaves like an Inspector, which stays open and displays live info from whatever object is currently selected.

Now, If we want change the fundamental behavior of the config dialog, we could do so, although I'm not sure if there's any good reason to do so.  Requiring an intentional action to commit changes would mean you couldn't close the dialog or switch components without answer a confirmation dialog if there were changes made.  In that case you'd probably want an Apply and OK button, where Apply doesn't close the dialog.  Perhaps if we were starting from scratch it might be a more straightforward approach, but I don't think it's really *better*, and given a userbase already accustomed to the current implementation, I think changing it now is just going to annoy a lot of people without bringing any significant benefit.

So my preference here would be to simply add a single "Revert" button, which restores the settings to where they were when the dialog was first opened.  I wouldn't favor using "Cancel" because that normally implies closing the dialog, and I don't think that "Revert" should close the dialog (it *could*, but it wouldn't be my choice).

PR #1957 (2023-01-08): [#960 & #1956] Add OK/Cancel button to component config & fix weird multi-comp edit bug
This PR fixes #960 by adding an OK and Cancel button to the component config dialog. I know that in PR #960 @neilweinstock expressed the desire for an alternative nomenclature, but I still want to stick with the "OK" and "Cancel" buttons simply because the texts are so dissimilar and short. If we were for instance to use "Cancel" and "Close", it would be easy for a user to misread e.g. "Cancel" for "Close". Or if you name it "Revert" and "Close", those are longer to read than simply "OK" and "Cancel".

Now  [... cut]
closed 2023-01-08 by JoePfeiffer