openrocket/openrocket #964: Jerky movement in 3D-view on Mac

neilweinstock, 2021-06-29:
I don't know when this crept in, but I just noticed it fairly recently.  Running Big Sur on an M1 Mac.

In 3D view, when rotating the rocket around by dragging the mouse, the movement is jerky.  It doesn't look like the frame rate is necessarily low, but more like there are several smooth frames then a pause, then repeat.  By comparison, the same operation in 15.03 is absolutely smooth.

I see this most but not all the time.  It may be that the window needs to be large enough to trigger this behavior.

Using the "apocalypse" sample rocket.

Here's 15.03:
https://user-images.githubusercontent.com/243872/123720784-e07ae380-d852-11eb-94fc-5a6f5f3920fa.mov

And here's the latest unstable:
https://user-images.githubusercontent.com/243872/123720808-f12b5980-d852-11eb-9bdd-94976ca3757b.mov

hcraigmiller, 2021-06-29:
Current unstable:
https://user-images.githubusercontent.com/68821492/123724106-9be64c80-d840-11eb-8186-84dd70ead4a7.mp4

[OS Windows 10 Pro service pack 120.2212.2020.0 with Java version 11.0.9 2020-10-20 LTS]

wolsen, 2021-06-30:
What Java are you running? An arm64 Java for m1 or an x86 version running through Rosetta?

There are some general reports around Java being significantly slower through Rosetta and I'm also curious if some of the jogl 3D libs need to be tuned for the m1 chip. 

These are just general thoughts and I'd be curious to see if there's a difference in Java runtimes. I'll double check we include arm builds of jogl native runtimes as well.

I have a raspberry pi4 I'll try to see if I can replicate some of the behavior on.

wolsen, 2021-06-30:
Will need an update to the jogamp/jogl libraries for Mac m1 arm74 support. I'll build a test build in the coming days with updated libs that folks can test on

neilweinstock, 2021-06-30:
I'll be excited to try that.

I'm running inside a packaged version that @itshanney provided a long time ago, not sure of the version.  I can guarantee it's an x86 binary, though.

I tried an ARM64 JDK I found on the web (can't remember which, might have been the AdoptOpenJDK) and the 3D stuff was totally broken, so I abandoned it quickly.

A good question to ask at some point is what Java to include in the packaged Mac version in the future.  Not sure if there's such a thing as a fat binary for java, or whether we should provide separate versions for x86 and ARM, or what.

wolsen, 2021-07-05:
@neilweinstock I got around to looking at the m1 support stuff today. The 3d libraries for jogl need to be updated. There's not an official, nightly or any build that includes support for this from the jogamp site. There's [this mail thread](http://forum.jogamp.org/JOGL-for-Mac-ARM-Silicon-td4040887.html) where Emmanuel from the Sweet Home 3D project rebuilt the native shared libraries and provided [a link](https://sourceforge.net/p/sweethome3d/code/HEAD/tree/tags/V_6_5_2/SweetHome3D/lib/java3d-1.6/macosx/) to their repositories for Java3D. The good news is that they are using the same jogl build we are, so I've downloaded those dynamic libraries and updated our macosx jar files to include them.

The build works for Linux, but I can't state whether or not these actually will work on your Mac. If you've got an aarch64/arm64 java version - care to give [this one](https://www.dropbox.com/s/37wrsumktxwmgcj/OpenRocket-mac-m1.jar?dl=0) a try?

neilweinstock, 2021-07-05:
I just tried it with AdoptOpenJDK form ARM64, v11.0.6.  The good news: 3D works.  The bad news: jerky movement is still apparent, along with the other anomalies I noted in other issues.

neilweinstock, 2021-07-06:
I will add to this: Photo Studio is *extremely* slow (both in this build and other unstable builds).

I can't help wondering if I'm really using a native Java, but as far as I can tell I am.

JoePfeiffer, 2021-07-06:
FWIW it is very responsive under Linux -- no noticeable lag at all.

neilweinstock, 2021-07-06:
I'll try to create a movie snippet tonight to illustrate what I'm seeing.

wolsen, 2021-07-08:
What settings do you have under the Graphics preferences? Does your experience change with tweaking any of the settings (Use Off-screen rendering, etc)?

neilweinstock, 2021-07-08:
Good point, I will experiment and report back.

neilweinstock, 2021-07-10:
> What settings do you have under the Graphics preferences? Does your experience change with tweaking any of the settings (Use Off-screen rendering, etc)?

Looks like you nailed it.  I had off-screen rendering checked (enabled).  When I disabled it, 3D operation became silky-smooth, both in the existing unstable running through Rosetta and the native ARM version.

I'm closing this issue.  Sorry for the false alarm!

closed 2021-07-10 by neilweinstock