openrocket/openrocket #965: Artifacts in 3D Finished view on Mac

neilweinstock, 2021-06-29:
Running latest unstable on an M1 Mac, Big Sur.

Here's what the "Apocalypse" sample rocket looks like in 3D finished view:
[image]

Note the areas in the red boxes.  In each case an inner coupler or shoulder is showing through and causing artifacts.  These do not appear in 15.03.

hcraigmiller, 2021-06-29:
Current unstable.

[image]
[image]

[OS Windows 10 Pro service pack 120.2212.2020.0 with Java version 11.0.9 2020-10-20 LTS]

neilweinstock, 2021-08-15:
I can no longer reproduce this on the latest unstable builds.

Since I'm the only on who ever saw it in the first place, I'm just going to close it now.

closed 2021-08-15 by neilweinstock