openrocket/openrocket #966: Strange behavior when clicking in 3D view on Mac

neilweinstock, 2021-06-29:
This is what happens when I click on the rocket in 3D view in the latest unstable, on M1 Mac running Big Sur:
https://user-images.githubusercontent.com/243872/123722157-c42c7600-d855-11eb-918e-e42b0cba2360.mov

Normally, clicking on the rocket in 3D view on the Mac does nothing: component selection hasn't worked for a long time (this is the subject of another issue).  Now, though, almost the whole rocket flashes when you click on it.  Or at least, maybe just the decal textures are flashing (hard to tell, but that's what it looks like to me).

hcraigmiller, 2021-06-29:
Current unstable.

The first left click selects the component, the second left click on the same component shades the entire rocket without opening the _configuration_ dialog box, and the third click (off the rocket or on a different component), removes the shading.

From the no component selected state, two quick left clicks on a component opens the _configuration_ dialog box AND shades the entire rocket. And, another left click (off the rocket or on a different component), removes the shading, leaving the _configuration_ dialog box open.

**Other than shading the entire rocket on the second left click of the same component** (any component), the program seems to run as expected.

[OS Windows 10 Pro service pack 120.2212.2020.0 with Java version 11.0.9 2020-10-20 LTS]

SiboVG, 2021-06-29:
I believe the cause of this problem is the way that selecting a component in the 3D view is programmed. That is: first all the components are rendered in a unique color and afterwards the selected color is translated back to know which component is selected.

Since this way if selecting an object does not work on Mac (see #754) and is in general somewhat janky (this is even mentioned in the code itself), I think the picking method itself should just be rewritten...

hcraigmiller, 2022-01-20:
I am no longer seeing the rendering issues that I described above. I just cannot select parts from the 3D views.

PR #1783 (2022-11-03): [#966] Fix weird behavior when clicking on textured part in 3D view
This PR fixes #966, which was a weird 3D behavior (on macOS, but maybe also others?) when clicking on a textured part (e.g. the body tube or fins of 'A simple model rocket').
closed 2022-11-03 by SiboVG