openrocket/openrocket #968: Bad highlight colors on Mac

neilweinstock, 2021-06-29:
This problem was present in 15.03, never got around to reporting it.  Currently running on an M1 Mac on Big Sur.

Certain UI elements are colored with a very dark highlight color, rendering the contained black text unreadable.  E.g.:
[image]

In the OS, when UI elements are highlighted with this color, the text is set to white to make it readable.

SiboVG, 2021-06-30:
Can confirm this problem on Macbook Air 2015, on Big Sur.

hcraigmiller, 2021-06-30:
Compare...

[image]

[OS Windows 10 Pro service pack 120.2212.2020.0 with Java version 11.0.9 2020-10-20 LTS]

PR #986 (2021-08-04): [fixes #968] Fix bad highlight colors
This PR fixes #968 for the most part, where some highlight colors or button colors were hardly readable.

The tables from the motor configuration now get assigned the correct foreground color (white) upon selection. This was fixed by changing the foreground in setSelected in FlightConfigurablePanel.

Also toggle buttons, such as the stage selector buttons, are fixed by adding and using a custom JToggleButton implementation, SelectColorToggleButton, which sets the foreground color correctly.

Finally the color of JButtons that are pressed is also fixed by a [... cut]
closed 2021-08-04 by neilweinstock