openrocket/openrocket #987: OpenRocket Crashes

roguextech, 2021-08-01:
I double-clicked the desktop shortcut, and a couple of seconds later, what you see in the screenshot is what happened.

I have attached the ORK file, user thrust curves and screenshot. The attached file is a 7zip file renamed to the .gif extension. I will be in and out this evening for about the next 4 hours. I am assuming that ya'll may want more info, I will get back with you.

JP

[image]

SiboVG, 2021-08-10:
Whilst I do not know the root cause of the issue, if you want to still keep using OR, you can temporarily move your user-defined thrust curves to somewhere else. I was able to recreate your issue when I moved the thrust curves that you included to my user thrust curves directory.

roguextech, 2021-09-07:
I figured out which file is causing the issue. RATT.eng is the culprit...

SiboVG, 2021-09-12:
Yup, indeed, you can also see inside the RATT.eng-file that there are indeed two thrust-values for a single time point, which is the thing that OR complains about...
[image]

The thing is, the error lies within the RATT.eng file and not really within OR itself; there shouldn't really be 2 or more thrust-values for a single time point.

What I can however do is show a pop-up window saying something in the line of "Thrust curve 'RATT.eng' has two thrust values for single time point, time[3]=0.278207, thrust=240.028; time[4]=0.278207, thrust=229.017. 'RATT.eng' will thus be ignored for importing the thrust curves. You can try replacing/deleting the thrust curve or manually editing it so that each time point has only one thrust value."

PR #1017 (2021-10-21): [fixes #987] Use warning dialog for thrust curve import issues
This PR fixes #987 where an issue in a user defined thrust curve file would throw an exception, thus crashing OR. This PR shows a warning dialog instead and ignores the faulty thrust curve file.

The warning dialog looks as follows:
[image]

Here is a [jar file](https://drive.google.com/file/d/1RGKhM87h_tnvKAoE2h72aREDLXgJyD-u/view?usp=sharing) for testing (note, you need to have [RATT.eng](https://github.com/openrocket/openrocket/files/7150168/RATT.eng.zip) in your user defined thrust curve directory to be able to test this issue, [... cut]
commit 5c6ca75e (2021-10-21): [fixes #987] Cleaner popup dialog message formatting
commit e3890712 (2021-10-21): [fixes #987] Use warning dialog for thrust curve import issues
commit ff523003 (2021-10-21): Merge pull request #1017 from SiboVG/issue-987

[fixes #987] Use warning dialog for thrust curve import issues