openrocket/openrocket #989: No 3D using Windows since last merge

hcraigmiller, 2021-08-05:
Since the last merge, I have lost all 3D functionality, both in the editor and in Photo Studio.

Latest merge:
[image]

OpenRocket-969-4.jar:
[image]

hcraigmiller, 2021-08-05:
Apparently, it is intermittent. When I selected _3D Unfinished_, the 3D functionality returned. I will test it more in depth and comment further,

hcraigmiller, 2021-08-05:
When I open the program and choose a _Simple model rocket_ from the example menu, the file loads without error, leaving the Rocket Configuration window open. If I then close that window and select _3D Finished_ view, 3D functionality is lost in the editor and Photo Studio. But, if I then select _3D unfinished_ view, 3D functionality returns.

The same is true if you start with any of the 3D views. It is not until the second 3D view is selected that 3D functionality returns.

ukhpr, 2021-08-05:
Same in Linux (in my case Arch, 
Also if you go to one 3D view, and then the other 3D view to make it appear, and then open another dialog window (eg debug log or bug report), it erases the 3D image.

Also the CPU load goes to 100%

The build I made on 2021-07-17 also works fine (both in Java 11 and 16)

SiboVG, 2021-08-07:
Hm, peculiar... I will make some builds of the latest commits to check after which one I may have messed some things up 🙈

SiboVG, 2021-08-08:
Since I cannot recreate the problem on my Mac, can someone test where the problems arises?

I have created 4 different builds from after the 2021-07-17 commit. OpenRocket-1 is the most recent one (the commit before the current unstable version), OpenRocket-2 the next most recent one etc. and OpenRocket-4 the oldest one, the one right after 2021-07-17.

So if someone could test out file OpenRocket-4 downto OpenRocket-1 and check from which commit the problem starts, that would help me track down the root cause.

[Here are the builds](https://www.dropbox.com/sh/whd9gfe4devuco9/AAAkawOyoRBuNvkh0-Mw7jfaa?dl=0).

ukhpr, 2021-08-08:
OpenRocket-4-295c926 seems to work OK on Linux with both openjdk-11 and -16.  I can't get any of the 3D stuff working at all on my Windows machine, but I think that's just my configuration, and unrelated.

The other 3 all immediately go to 100% CPU load on 2 cores, simply starting the application, before even loading a rocket.  All 3 have the issue noted above, where you need to select a second 3D view to see anything, and any popover window erases the 3D image without it being redrawn.

hcraigmiller, 2021-08-08:
Okay, **OpenRocket-969-4.jar works, as does your 4**. But, your 1-3 do not work.

SiboVG, 2021-08-10:
Thanks for testing @ukhpr  and @hcraigmiller! Working on a fix right now.

I've run some more tests on my Windows 10 Home and noticed that I was able to recreate the blank editor window by resizing the window, even with much older releases. If someone else experienced this problem, maybe there should be a new issue to be made about 3D view loss upon window resizing?

ukhpr, 2021-08-10:
I'm assuming that issue #992 is related to #989 ? Hope the suggested fix there works.

SiboVG, 2021-08-10:
> I'm assuming that issue #992 is related to #989 ? Hope the suggested fix there works.

Yes, it is, the foreground setting should never have been inside the pain-method, my mistake. I'm looking for a way to still get the desired functionality while having it outside of the paint-method.

hcraigmiller, 2021-08-10:
This appears to have fixed the issues... no anomalies found.

[OS Windows 10 Pro service pack 120.2212.2020.0 with Java version 11.0.9 2020-10-20 LTS]

ukhpr, 2021-08-10:
Looks good to me in Linux with both OpenJDK 11.0.12 and 16.0.2
I can't reproduce any of the issues noted above with the 989 jar file

PR #993 (2021-08-15): [fixes #989 & #992] Fix 3D view going blank + high CPU usage
This PR (I do very much hope) fixes #989 and #992 where the editor view went blank when switching to the 3D view + where a very high increase in CPU usage was noticed. Okay, so I did an oopsie in my PR of fixing bad highlight colors. The code written to change the foreground for the SelectColorButton and SelectColorToggleButton should never have been put inside the paint-method. Not only does this cause a very high CPU usage, it also messes up the init-method calling of the editor window (which caused it to go blank).

I added the fo [... cut]
commit 2f6a8c8d (2021-08-15): [fixes #989] Use changelistener for foreground change + don't change on non OSX-devices
commit 4d8e4be6 (2021-08-15): Merge pull request #993 from SiboVG/issue-989

[fixes #989 & #992] Fix 3D view going blank + high CPU usage
commit 716e723a (2021-08-15): [fixes #989] Fix toggleButton going 'blank' upon window out of focus
closed 2021-08-15 by wolsen