openrocket/openrocket #992: Very high CPU usage

ShadowNinja, 2021-08-10:
After a recent update I found the app consuming about 200% CPU (100% in the java process + 100% in the display server process).  I bisected the issue to commit ebcdd469b7678d1b96202634b97ad54077474af8 (Use SelectColorToggleButton).

It looks like `SelectColorToggleButton` resets the foreground style on every frame.  Changing the code to only call `setForground` when the selected state has actually changed resolved the issue, see below:

```diff
--- a/swing/src/net/sf/openrocket/gui/widgets/SelectColorToggleButton.java
+++ b/swing/src/net/sf/openrocket/gui/widgets/SelectColorToggleButton.java
@@ -45,13 +45,13 @@ public class SelectColorToggleButton extends JToggleButton {
     }
 
     @Override
-    public void paint(Graphics g) {
-        if (isSelected()) {
+    public void setSelected(boolean selected) {
+        if (selected) {
             setForeground(UIManager.getColor("ToggleButton.selectForeground"));
         }
         else {
             setForeground(UIManager.getColor("ToggleButton.foreground"));
         }
-        super.paint(g);
+        super.setSelected(selected);
     }
 }
```

SiboVG, 2021-08-10:
Indeed, the foreground setting should never have been inside the paint-method, my mistake. However, placing it inside the setSelected-method does not yield the desired functionality of setting a white foreground on the blue background on OSX-systems when the button is selected.

SiboVG, 2021-08-10:
The issue might be resolved in #993, can you check?

ShadowNinja, 2021-08-12:
Yep, #993 fixes the issue.

closed 2021-08-15 by wolsen