openrocket/openrocket #996: Feature request: better file dialog on Mac

neilweinstock, 2021-08-16:
The Mac file dialogs are incredibly minimal, moreso than for any other app I can think of.  Here's the OR open file dialog:
[image]
By contrast, here's the open file dialog on Preview.app:
[image]
Even if the most full-featured file dialog is not possible for some reason in Java, my hope would be that *something* better than what's there now would be possible.

The Windows File dialog is also minimal compared to other apps, but it is quite a bit better than the Mac dialog.  If someone wants to upgrade the Windows dialog too, I won't complain. :)

neilweinstock, 2021-08-16:
[JavaFX FileChooser](https://docs.oracle.com/javafx/2/ui_controls/file-chooser.htm) looks pretty decent...  are we able to use that, or are we restricted to Swing controls?

derreisende77, 2021-09-08:
On macOS and windows you may also use java.awt.FileDialog in FileDialog.LOAD mode with a JFrame parent to get the above shown native file dialog. I am using this successfully in my mac and windows java app.

jimmiedave, 2022-09-23:
Current minimal MacOS File Dialog does not follow aliases/shortcuts to folders. If I have a shortcut on my desktop, pointing to a folder deep in the disk, I can't access it with the OR file dialog, where I can do so with any MacOS dialog.

jimmiedave, 2024-06-02:
While we're batting this around (wake up, feature request!) It'd be good if the dialog followed OS-native conventions on tabbing, shortcuts and keyboard accelerators. I guess I'm mainly asking for native platform file dialogs. Would be better on Mac, on Windows, and...I'm not sure on Linux - not certain Linux has settled down on a single way.

neilweinstock, 2024-06-02:
There's some good info here:  https://stackoverflow.com/questions/39819319/windows-native-file-chooser-in-java

A promising solution is https://github.com/steffen678/NativeJFileChooser.  I don't know what it takes to run JavaFX widgets under Swing, but there is a lot written about that around the web.

closed 2026-10-03 by neilweinstock