openrocket/openrocket #997: Photo Studio flame and spark Effect no longer display correctly in Windows 10 Pro

hcraigmiller, 2021-08-21:
Beginning with OpenRocket-954-jkl1.jar dated 07/02/2021, and post 07/11/2021 merge, the Photo Studio flame and spark effects no longer display correctly in Windows 10 Pro.
[image]

A similar error may be what is causing the smoke effect not to function as well (see Issue #874).

SiboVG, 2021-08-23:
So the issues are related to #967? Do you also have an idea of a merge date after which #874 started showing up?

hcraigmiller, 2021-08-24:
I don't. My records only go back to the 11/03/2020 merge and it existed from and after that date, but I don't know if it existed before that date.

hcraigmiller, 2022-01-11:
This bug occurs **only if there is more than one stage**, even if the other stages have no components, **and then only if the motor is NOT in the upper most stage**. 

For example:
   1. If Upper Stage then Motor Stage then Flame is a glow not a flame.
   2. If Motor Stage then Lower Stage the Flame is a flame not a glow.
  
Regardless of how may stages there are, there must be a motor in the upper stage for the Flame to be a flame instead of a glow. **This is the opposite of what it should be**; if the lowest stage has a motor, the Flame should be a flame, not a glow.

The non-functional smoke issue is separate and unrelated.

SiboVG, 2022-01-11:
> This bug occurs only if there is more than one stage, even if the other stages have no components, and then only if the motor is NOT in the upper most stage.

Aha! At last I am able to recreate this issue. Indeed, I tested it with 'A simple model rocket' and with 'Three-staged rocket'. The former displays the flame and sparks, the latter not. If you look closely at the image, you see that the smoke of 'Three-staged rocket' starts somewhere in the middle of the rocket. If you also increase the aspect ratio of the flame, eventually it pokes out of the bottom of the rocket.
[image]

So the starting point of the smoke, flame and sparks is not at the right motor.

PR #1069 (2022-01-12): [fixes #997] Fix Photo Studio exhaust effects starting at top stage
This PR fixes #997. The issue was that the exhaust effects' start point was at the top stage, as can be [seen here](https://github.com/openrocket/openrocket/issues/997#issuecomment-1010332110), instead of on the bottom stage. The fix was very simple, because there was already logic implemented to use the bottom stage motor for the exhaust effects, but the problem was that there was a bottomStage.getStageNumber() method that was never assigned to the bottomStageNumber variable, causing it to always remain 0. This PR just simply [... cut]
commit 2184cc80 (2022-01-12): [fixes #997] Update bottomStageNumber
The bottomStage.getStageNumber() was called, but never assigned to bottomStageNumber. Now it is. This has the effect that the exhaust effects will start at the bottom stage instead of on the top stage, which was previously the case.
commit 7fdd0ed5 (2022-01-12): Merge pull request #1069 from SiboVG/issue-997

[fixes #997] Fix Photo Studio exhaust effects starting at top stage
closed 2022-01-12 by JoePfeiffer